linux-nvme/nvme-cli #2: Little-Big Endian

keithbusch, 2015-02-13:
I know it's crazy, but I hear some people don't use x86. I don't want those folks to feel dejected, so I'll post the issue so they know the little people are looking out for the big people (pun intended).

sbates130272, 2015-06-02:
There is an issue compiling for ppc64el as of 62bbb67caf757e4f41141aebd24ebb7850ce6c8e.

batesste@ubuntu-ppc:~/nvme-cli$ make nvme
cc -c -I  -m64 -std=gnu99 -O2 -g -pthread -D_GNU_SOURCE -D_REENTRANT -Wall -Werror ./src/argconfig.c
cc -c -I  -m64 -std=gnu99 -O2 -g -pthread -D_GNU_SOURCE -D_REENTRANT -Wall -Werror ./src/suffix.c
cc -I  -m64 -std=gnu99 -O2 -g -pthread -D_GNU_SOURCE -D_REENTRANT -Wall -Werror nvme.c -lm -o nvme argconfig.o suffix.o
nvme.c: In function ‘show_error_log’:
nvme.c:167:3: error: format ‘%lld’ expects argument of type ‘long long int’, but argument 2 has type ‘__u64’ [-Werror=format=]
   printf("error_count  : %lld\n", err_log[i].error_count);
   ^
nvme.c:172:3: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 2 has type ‘__u64’ [-Werror=format=]
   printf("lba          : %#llx\n", err_log[i].lba);
   ^
nvme.c: In function ‘show_nvme_resv_report’:
nvme.c:195:3: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 2 has type ‘__le64’ [-Werror=format=]
   printf("  hostid  : %llx\n", le64toh(status->regctl_ds[i].hostid));
   ^
nvme.c:196:3: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 2 has type ‘__le64’ [-Werror=format=]
   printf("  rkey    : %llx\n", le64toh(status->regctl_ds[i].rkey));
   ^
nvme.c: In function ‘show_fw_log’:
nvme.c:222:7: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘__u64’ [-Werror=format=]
       fw_to_string(fw_log->frs[i]));
       ^
nvme.c: In function ‘show_lba_range’:
nvme.c:753:3: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 2 has type ‘__u64’ [-Werror=format=]
   printf("slba       : %#llx", lbrt[i].slba);
   ^
nvme.c:754:3: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 2 has type ‘__u64’ [-Werror=format=]
   printf("nlb        : %#llx", lbrt[i].nlb);
   ^
nvme.c: In function ‘show_nvme_id_ns’:
nvme.c:903:2: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 2 has type ‘__le64’ [-Werror=format=]
  printf("nsze    : %#llx\n", ns->nsze);
  ^
nvme.c:904:2: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 2 has type ‘__le64’ [-Werror=format=]
  printf("ncap    : %#llx\n", ns->ncap);
  ^
nvme.c:905:2: error: format ‘%llx’ expects argument of type ‘long long unsigned int’, but argument 2 has type ‘__le64’ [-Werror=format=]
  printf("nuse    : %#llx\n", ns->nuse);
  ^
cc1: all warnings being treated as errors
Makefile:25: recipe for target 'nvme' failed
make: *** [nvme] Error 1

keithbusch, 2016-03-07:
I think all the le*toh's are in place now. Will close this.

closed 2016-03-07 by keithbusch