linux-nvme/nvme-cli #5: nvme fails when local nvme.h mismatches actual kernel version

sbates130272, 2015-02-20:
Hi

Right now the CLI compiles against a local copy of the uapi nvme.h file. This copy, of course, may differ, from the one the kernel (and nvme module if applicable) were compiled against. This can lead to all sorts of fun and games when we pass a structure we populated with the local header into the kernel IOCTL.

We need to find a better way of handling this. Usually the actually header the kernel was compiled with will be located at /lib/modules/$(uname -r)/build/include/uapi so maybe our Makefile should test for the existence of this file and use it when available and use the local file as a last resort. Or maybe we use a call to uname -r to bork if needed in the make process if we suspect the local header and kernel differ too much?

Stephen

keithbusch, 2015-02-20:
I believe if the kernel's uapi headers are install correctly, they are usually put in the include path at /usr/include/linux/. YMMV.

Maintaining ABI with user space is pretty important though. We can count on existing ioctl structures to never change.

sbates130272, 2015-03-15:
I have an initial fix for this in c0d4544d4bc6648f8168f62d8846fb345e959e6f. Which tries to use the uapi for the running kernel at build time. It is not perfect so keeping this open for now but a step up from where we were before.

keithbusch, 2017-05-03:
I think we can close this one. The nvme-cli hasn't included the kernel's uapi nvme header in a long time.

closed 2017-05-03 by keithbusch