linux-nvme/nvme-cli #9: Cut and paste error disables passthru command

jimmgrogan, 2015-05-01:
The the file nvme.c in the function static int nvme_passthru(int argc, char **argv, int ioctl_cmd) at line 2248 there is a typo assigning the cdw13 value to multiple fields. This makes many vendor specific commands unusable.

	cmd.cdw2         = cfg.cdw13;
	cmd.cdw3         = cfg.cdw13;
	cmd.cdw10        = cfg.cdw13;
	cmd.cdw11        = cfg.cdw13;
	cmd.cdw12        = cfg.cdw13;
	cmd.cdw13        = cfg.cdw13;
	cmd.cdw14        = cfg.cdw14;
	cmd.cdw15        = cfg.cdw15;
	cmd.opcode       = cfg.opcode;
	cmd.flags        = cfg.flags;
	cmd.rsvd1        = cfg.rsvd;
	cmd.nsid         = cfg.namespace_id;

sbates130272, 2015-05-01:
Thanks! I will fix this one later today.

Stephen

sbates130272, 2015-05-01:
Fixed in baac1cafee73a5a8b0e3628044146d13d70816e0.

closed 2015-05-01 by sbates130272

jimmgrogan, 2015-05-01:
Thanks

Fixed in baac1ca<https://github.com/linux-nvme/nvme-cli/commit/baac1cafee73a5a8b0e3628044146d13d70816e0>.