linux-nvme/nvme-cli #18: mismatch argconfig_type in submit_io()

DaisukeIwai, 2015-07-01:
`regress` issues read/write command with --data_size=4k, 
however argconfig_type in submit_io()'s data_size is "CFG_POSITIVE".
I guess either 
* replace CFG_POSITIVE as CFG_LONG_SUFFIX in nvme.c line 2608
or
* replace RAND_SIZE=4k as RAND_SIZE=4096 in regress line 29
may be applied.

c8aab9676058e1faea756a1626ef04de65a3cf44

sbates130272, 2015-07-01:
I would prefer to fix this in nvme.c. Let me see why this was never spotted before now and get back to you, Thanks for raising it as an issue. Does this break regress for you? If so can you tell me something about your system?

sbates130272, 2015-07-02:
OK fixed along with a nasty bug where submit_io always returned 0! Thanks.

closed 2015-07-02 by sbates130272

DaisukeIwai, 2015-07-02:
Thanks for reply & quck fix. Yes, `regress` with c8aab9676058e1faea756a1626ef04de65a3cf44 breaks with "free(): invalid next size (fast)" in submit_io(). --data_size=4k was equal to --data_size=4, only 4byte size of buffer is not enough for read data buffer. Passed with f579aa4a23b338edbe298c98bbaada47074847b8 on Ubuntu 15.04 x86_64 and our NVMe device. thank you,