linux-nvme/nvme-cli #20: getopt_long_only needs 'int' for integer value.

DaisukeIwai, 2015-07-05:
It seems CFG_NONE and no_argument type argument may corrupt other parameters by get_opt_long().
nvme.c uses '__u8' for CFG_NONE&no_argument type, however getopt_long_only() in argconfig_parse() assume it as 'int'.
__u8 argument may be overwritten with 32bit value, and corrupts other parameter value.
These __u8 member should be modified with __u32. Checked on Ubuntu 15.04 x86_64.

For example, "--raw-binary" and "--read" are CFG_NONE&no_argument type arguments.
$ nvme io-passthru /dev/nvme0 --opcode=2 --namespace-id=1 --data-len=4096 --cdw10=0 --cdw11=0 --raw-binary --read
Pass: OK
$ nvme io-passthru /dev/nvme0 --opcode=2 --namespace-id=1 --data-len=4096 --cdw10=0 --cdw11=0 --read --raw-binary
Fail: "data direction not given" error

Could you please check changes in e14b613926e91e0105c97b8734f23bbc7169c1c3 ?

sirosiro, 2015-07-05:
I also think so. It will be happen the same problem to on the LP64 and LLP64 except of ILP64 memory model.

sbates130272, 2015-07-05:
Hi

Thanks for this. Can you submit this as a pull request? You might need to rebase it after we merge another PR we have staged (which should happen in the next day or two).

Cheers

Stephen

keithbusch, 2015-10-20:
Closing as I believe this is fixed in the most current version.

closed 2015-10-20 by keithbusch