linux-nvme/nvme-cli #22: Facing problem in using the command mentioned in the list

Rajnikantcmr, 2015-07-20:
Hi Guys
  i am facing problem in uses of the command mentioned in list:
[image]
i am able to use the first command but others i can not.i had done all the setup which required for using NVMe-Cli like compiling kernel version 3.19 and others done.just need the syntax to execute these command.can anyone help me.
Thanks
Rajnikant

keithbusch, 2015-07-20:
In your example, "id_ns" is not a sub-command. You're looking for "id-ns" (dash instead of underscore), so it should look like this:

  # nvme id-ns /dev/nvme0n1

Rajnikantcmr, 2015-07-20:
Hi Keith
 it was silly mistake of me.now it is working.Thank you for your guidance.
i want to know,can we run this complete sequence at a time.and i have one
more qurey does this test framework tell about the SSD Serial#.
  Will be waiting for your reply.
Thanks
Rajnikant

keithbusch, 2015-07-20:
The serial number comes from 'id-ctrl'.

Rajnikantcmr, 2015-07-20:
>  [image: Boxbe] <https://www.boxbe.com/overview> This message is eligible
> for Automatic Cleanup! (notifications@github.com) Add cleanup rule
> <https://www.boxbe.com/popup?url=https%3A%2F%2Fwww.boxbe.com%2Fcleanup%3Ftoken%3DF1Ohe7XyjMTqMmx7Iass8CgLJOe6w3uNAqr%252FC9Y99JXlw7q1sTOrN8%252B3YegslQDSkMSE8gE0FFrDyhrROvTc4Tl4hKFk2S62r7r%252F2vnz%252Fw2X%252F8BtFZY6Af0XJGehRklGH3uteykYwcjdQifjvZ0aeQ%253D%253D%26key%3D2R7US2TQiFOa3WCgVERpzzeEMth1NueOHQTpWlqHTVA%253D&tc_serial=22031010924&tc_rand=1403415659&utm_source=stf&utm_medium=email&utm_campaign=ANNO_CLEANUP_ADD&utm_content=001>
> | More info
> <http://blog.boxbe.com/general/boxbe-automatic-cleanup?tc_serial=22031010924&tc_rand=1403415659&utm_source=stf&utm_medium=email&utm_campaign=ANNO_CLEANUP_ADD&utm_content=001>
>
> The serial number comes from 'id-ctrl'.

Hi Keithbusch
 i did not got answer for my previous question regarding run all of the
command together.
Thanks
Rajnikant

keithbusch, 2015-07-20:
I'm not sure I understand the question. Are you wanting to execute multiple sub-commands in a single invocation of the program? I suppose we could do that, or some kind of interactive mode.

Rajnikantcmr, 2015-07-20:
Hi Keithbusch
   Yes i want to execute multiple sub-commands in a single invocation of
the program and save the results on system.

Thanks
Rajnikant

keithbusch, 2015-07-20:
That's not a feature of this program (yet). I'm sure you can script these things together to get somewhat similar observable output, though.

Rajnikantcmr, 2015-07-21:
Thank you for your time and valuable guidelines.

closed 2015-08-21 by keithbusch