linux-nvme/nvme-cli #27: make fails with 'error: format not a string literal and no format arguments [-Werror=format-security]'

fromscratch-xx, 2015-08-21:
I am trying to build nvme-cli on latest Ubuntu 14.04 trusty, but it fails with:
nvme-cli$ make
  cc -c -I   -m64 -std=gnu99 -O2 -g -pthread -D_GNU_SOURCE -D_REENTRANT -Wall -Werror ./src/argconfig.c
./src/argconfig.c: In function ‘argconfig_print_help’:
./src/argconfig.c:161:5: error: format not a string literal and no format arguments [-Werror=format-security]
     printf(buf);
     ^
cc1: all warnings being treated as errors
make: *** [argconfig.o] Error 1

I replaced the printf with
*fputs(buf, stderr);
and the make runs through. Maybe you can take a look at that!

All the best, Georg

keithbusch, 2015-08-21:
thanks for the notice. The gcc version tested with apparently doesn't complain, but need to fix. Maybe instead do: printf("%s", buf)

keithbusch, 2015-08-21:
Fixed now. Sorry I wasn't able to properly attribute your report in the change log.

closed 2015-08-21 by keithbusch