linux-nvme/nvme-cli #28: log entries default value for nvme-error-log command

samiWaheed, 2015-08-27:
nvme error-log command has a default of 64 entries (--log-entries).  This default value is causing errors on some devices that do not support 64 entries.

I would like the default value for log entries set to the maximum supported by the device.  This is reported in "Error Log Page Entries (ELPE)" field in the ID controller data structure.

keithbusch, 2015-08-27:
Oof, well, if you can't be bothered to write it in manually, we've reach a bit of a limitation here. We can't open the device to query ID controller until after the arguments are parsed because we don't know where in the command line you've specified your device, but the argument parsing requires we already set the defaults. This is a chicken and egg situation.

jderrick, 2015-08-27:
To expand on Keith, we have quite a few instances where input would need to
be validated. For example, there are plenty of ways to do 'nvme read'
incorrectly. Either we query the controller/ns before and validate, or we
force the user to do it correctly

samiWaheed, 2015-08-27:
I have no problem writing it manually and I already do so in the scripts that I use.  Wanted to bring it up to your attention in case you care!

Also, looking at your code, what is wrong with doing something like the following:
```
static int get_error_log(int argc, char **argv)
{
.
.
.
        struct config cfg;
        //*** START of change
        struct nvme_id_ctrl ctrl;
        int entries=64;
        get_dev(1, argc, argv);
        err = identify(0, &ctrl, 1);
        if (!err) {
           entries = ctrl.elpe+1;     // Zero based counter
        }
        const struct config defaults = {
                .namespace_id = 0xffffffff,
                .log_entries  = entries,
        };
       //*** END
       const struct argconfig_commandline_options command_line_options[] = {
```

keithbusch, 2015-08-28:
I do think it'd be a handy feature, but just saying the ordering of events means we can't pre-determine the desired default. The device to open is the first non-argument parameter (after the sub-command). We have no idea where in the argument list this is until we actually parse them. What you're proposing happens to work with something like this:

`  # nvme error-log /dev/nvme0`

But fails with:

`  # nvme error-log -e 8 /dev/nvme0 -n 1`

Which should be perfectly fine with standard gnu options.

We could instead override the chosen setting to `min(device's max, cfg.default)` after the arguments were parsed with the existing default.

keithbusch, 2015-08-28:
Fixed here: https://github.com/linux-nvme/nvme-cli/commit/5018e560ba537020fd49f7b9b7df3e8aaa99e9f7

closed 2015-08-28 by keithbusch

samiWaheed, 2015-08-28:
@keithbusch Fix works.  Thanks.