linux-nvme/nvme-cli #35: Build errors on i686

ajorg, 2015-09-28:
We're not using 32-bit right now, but here is the output from a build attempt:
```
cc -I -m64 -std=gnu99 -O2 -g -pthread -D_GNU_SOURCE -D_REENTRANT -Wall -Werror -DLIBUDEV_EXISTS nvme.c -lm -ludev -o nvme argconfig.o suffix.o
cc1: warnings being treated as errors
nvme.c: In function 'nvme_attach_ns':
nvme.c:1535: error: cast from pointer to integer of different size
nvme.c: In function 'nvme_feature':
nvme.c:2114: error: cast from pointer to integer of different size
nvme.c: In function 'fw_download':
nvme.c:2317: error: cast from pointer to integer of different size
nvme.c: In function 'sec_send':
nvme.c:2707: error: cast from pointer to integer of different size
nvme.c: In function 'resv_acquire':
nvme.c:2849: error: cast from pointer to integer of different size
nvme.c: In function 'resv_register':
nvme.c:2943: error: cast from pointer to integer of different size
nvme.c: In function 'resv_release':
nvme.c:3038: error: cast from pointer to integer of different size
nvme.c: In function 'resv_report':
nvme.c:3120: error: cast from pointer to integer of different size
nvme.c: In function 'submit_io':
nvme.c:3270: error: cast from pointer to integer of different size
nvme.c:3272: error: cast from pointer to integer of different size
nvme.c:3279: error: cast to pointer from integer of different size
nvme.c:3280: error: cast to pointer from integer of different size
nvme.c:3281: error: cast to pointer from integer of different size
nvme.c: In function 'sec_recv':
nvme.c:3401: error: cast from pointer to integer of different size
nvme.c: In function 'nvme_passthru':
nvme.c:3569: error: cast from pointer to integer of different size
nvme.c:3571: error: cast from pointer to integer of different size
nvme.c:3581: error: cast to pointer from integer of different size
nvme.c:3596: error: cast to pointer from integer of different size
nvme.c:3597: error: cast to pointer from integer of different size
nvme.c:3614: error: cast to pointer from integer of different size
nvme.c:3616: error: cast to pointer from integer of different size
```

keithbusch, 2015-09-28:
Okay, we'll have to do some tricks; these are all safe.

keithbusch, 2015-10-20:
Fixed in commit:

https://github.com/linux-nvme/nvme-cli/commit/992d220362e0e64f2c38a24cc6357b2121baf484

closed 2015-10-20 by keithbusch