linux-nvme/nvme-cli #47: debian changelog should be auto-magically generated

sbates130272, 2016-01-11:
Hi

Commit 15878d3d285c126b10c51f854748bcd6f5be2383 added a debian/changelog file. This breaks the Ubuntu PPA make flow since we auto-magically generate that file. I would rather not have this file in the repo. If you want you can have a changelog.debian file in the repo and copy it over when you build your package but a better option is to add echo commands to the Makefile like we did for deb-ppa.

Discuss ;-)

Stephen

leitao, 2016-01-11:
Hello Stephen,

Why does it break the PPA flow? Also, how do you generate the changelog file for the PPA? 
I am used to have the changelog files being kept from previous release and updated for new releases.

sbates130272, 2016-01-14:
Hi Breno

Sorry for the delay. The issue is your change adds the changelog file to the repo. Since we auto-gen it we overwrite what you have and someone will forget to undo that and check it in by mistake. I'd prefer it if you either auto-gen in the Makefile like we do (you can use Macro's to swap in your text) or have a copy somewhere else and copy it into the debian folder. Either way we should keep changelist out of the repo and in the .gitignore list.

Stephen

sbates130272, 2016-01-22:
Hi Breno

Are you planning to submit a PR to address this? If not I can implement a fix based on a copied over file...

Stephen

leitao, 2016-01-22:
HI Stephen,

I am working with the Debian community to have the fixes they want done,
and once the patch gets accepted, I will submit a PR.

THe discussion is happening at
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=810033.

But, from what I understand, the changelog should not be generated
automatically. Mainly because debian/changelog tracks the package changes,
not the upstream changes itself, so, it is not easy to generate it
automatically, and, there is no way to import it from git.

sbates130272, 2016-01-25:
Hi Breno

OK let's wait until your PR comes in. We can live with your change but my concern is that when I do a Ubuntu PPA .deb I auto-create my chanelog file and if I forget and later do a git commit -a I am going to submit my temporary changelog on top of your file and smush it. We have a couple of other people doing Debian/Ubuntu packaging for internal purposes and they will see similar issues.

Any thoughts on how we address that?

Stephen

leitao, 2016-02-29:
Stephen,

Nvme-cli version 0.3 is now part of Debian, and made available in almost all architectures.

https://buildd.debian.org/status/package.php?p=nvme-cli&suite=sid
So, a PPA is not required anymore, unless there is changes on the package that is still not upstream yet.

sbates130272, 2016-02-29:
Breno

Great work. As I understand it this means that Ubuntu users also get access via the "Universe" packages. Do I have that correct? How do we trigger an update to the package if we want to include new features?

Stephen

igaw, 2022-01-19:
No progess in this bug report for a long time. If you think it's still relevant please reopen.

closed 2022-01-19 by igaw