linux-nvme/nvme-cli #59: add autoconf/autotools ?

pmmccorm, 2016-02-26:
We all loathe autotools for good reasons, but it is does provide some nice features that nvme-cli will want or already implements itself:

1) easy cross compilation
2) configurable locations for everything (prefix, destdir, sysconfdir)
3) detection of libraries (libudev), headers (nvme.h or nvme-ioctl.h), and struct members
4) handles version substitution in source files

Out of all these I think number three can provide us some real benefits: we can generate defines for certain fields in struct nvme_id_ctrl, and what ioctls are available (eg use for #57).

Is this something people are interested in? I can issue a PR if so.

sbates130272, 2016-02-26:
I'd been keen to see the PR.

keithbusch, 2016-02-26:
I think this is a good idea.

pmmccorm, 2016-03-02:
This PR is coming. That argconfig refactor broke my world, rebasing.