linux-nvme/nvme-cli #62: Release package is being generated without version file

leitao, 2016-03-01:
The packages being release, as 0.3, 0.4 do NOT contain the version file, and it is not also part of a git archive, which means that the version shows up as 0.1

This is one of the problem we have at Ubuntu and Debian at the moment. 

I suggest that we generate a version file and update it before each release commit, and make the git tag on top of this commit.

keithbusch, 2016-03-01:
Thanks for the notice. You're right, the tag should have been set after updating the default version file. I've pushed the update, and will see if we can auto-generate this in the future so the mistake doesn't happen again.

pmmccorm, 2016-03-02:
The autotooling might help with this. Why is this file needed? And does it belong in the "dist" tarball?

keithbusch, 2016-03-02:
The file is used to set a version in the code, copied straight from my other favorite tools: git and fio.

leitao, 2016-03-02:
I am uploading the new 0.4 version to Debian and Ubuntu, and I just fix this using a hack. It would be good if we have it fixed in the next release.

This is version 0.4 at Debian mentors:
http://mentors.debian.net/debian/pool/main/n/nvme-cli/nvme-cli_0.4-1.dsc

keithbusch, 2016-03-14:
v4 was pretty busted from some half-baked ideas that weren't well tested. I think we've settled all the regressions at this point and will tag a new release.

closed 2016-05-11 by keithbusch