linux-nvme/nvme-cli #68: Only keep one set of documentation in the git repo

pmmccorm, 2016-03-04:
There are currently three copies of each command documentation under Documentation: man page, html, and text.

The packages only need the man pages, do we need to keep the other formats in the tree? Can we just generate those as needed? What are the consumers of the .txt and html formats?

keithbusch, 2016-03-05:
I don't speak manpage. But I can speak asciidoc, which are is how the txt files are written. The html are generated for free, and I share these with end users.

The only reason to keep them in source control is because the majority of users don't have asciidoc installed, so they can't generate the artifacts from the documentation.

pmmccorm, 2016-03-07:
Oh I see, so the man pages are really the build artifacts-- but if we removed those the build would require asciidoc as well.

closed 2016-03-07 by keithbusch