linux-nvme/nvme-cli #69: Send IO transaction for 2 sectors.

ramyakanth, 2016-03-07:
Hi,  I have been able to do IO transaction for single sector i.e., 512B, but when I try to for 1024Bytes with Protection Information enabled,   it does not seem to work and I couldn't find much information in the document. can you please send me the exact command to send data of 1024Bytes with End to End protection enabled ?, thanks. 

-Ram.

keithbusch, 2016-03-07:
Need to verify your the physical format, metadata size, pi settings, and metadata location to know how to construct the command. Could you send send the output from `nvme id-ns /dev/nvme...`?

ramyakanth, 2016-03-07:
Hi Keith,
Please check the requested information below, thanks

NVME Identify Namespace 1:
nsze    : 0x42428000
ncap    : 0x42428000
nuse    : 0x1
nsfeat  : 0
nlbaf   : 3
flbas   : 0x1
mc      : 0x3
dpc     : 0x1f
dps     : 0x3
nmic    : 0
rescap  : 0x7f
fpi     : 0
nawun   : 0
nawupf  : 0
nacwu   : 0
nabsn   : 0
nabo    : 0
nabspf  : 0
nvmcap  : 0
nguid   : 00000000000000000000000000000000
eui64   : 04e00002fdf94801
lbaf  0 : ms:0   ds:9  rp:0x3
lbaf  1 : ms:8   ds:9  rp:0x3 (in use)
lbaf  2 : ms:0   ds:12 rp:0
lbaf  3 : ms:8   ds:12 rp:0

keithbusch, 2016-03-07:
Okay, 512b block, 8b metadata as a separate buffer, pi type 3.

First, please note that the nvme-cli is not going to allocate and generate/verify your integrity payload. You need to do that yourself.

Second, not every version of the Linux kernel handles metadata setup correctly. The most recent ones should handle this.

The following assumes your data is in a file called "data" and your metadata integrity payload is in a file called "metadata". If you want to write 2 blocks of data with 16 meta-data integrity payload:

`nvme write /dev/nvme0n1 --start-block=0 --block-count=1 --data-size=1024 --metadata-size=16 --data=data --metadata=metadata --prinfo=4`

You can clear prinfo if you want to disable the guard tag checking while figuring out how to construct the metadata payload.

keithbusch, 2016-03-14:
Any update, or can we close this issue?

ramyakanth, 2016-03-15:
Thanks,  Keith.  We can close this issue.

closed 2016-03-15 by ramyakanth