linux-nvme/nvme-cli #73: Compilation errors on 32-bit architecture

RussellJoyce, 2016-03-14:
When running `make` on 32-bit ARM, the following compile errors are caused by casts from `void*` to `__u64`. This did not happen in v0.3, but does from v0.4 onwards, I assume due to the removal of the `unsigned long` casts that used to be present.

`NVME_VERSION = 0.4.31.ge999
cc -I ./src -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.4.31.ge999"' -c ./src/argconfig.c
cc -I ./src -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.4.31.ge999"' -c ./src/suffix.c
cc -I ./src -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.4.31.ge999"' -c nvme-print.c
cc -I ./src -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.4.31.ge999"' -c nvme-ioctl.c
nvme-ioctl.c: In function ‘nvme_passthru’:
nvme-ioctl.c:68:15: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .metadata = (__u64) metadata,
               ^
nvme-ioctl.c:69:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) data,
            ^
nvme-ioctl.c: In function ‘nvme_io’:
nvme-ioctl.c:99:15: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .metadata = (__u64) metadata,
               ^
nvme-ioctl.c:100:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) data,
            ^
nvme-ioctl.c: In function ‘nvme_dsm’:
nvme-ioctl.c:191:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) dsm,
            ^
nvme-ioctl.c: In function ‘nvme_resv_acquire’:
nvme-ioctl.c:225:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) (payload),
            ^

nvme-ioctl.c: In function ‘nvme_resv_register’:
nvme-ioctl.c:242:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) (payload),
            ^
nvme-ioctl.c: In function ‘nvme_resv_release’:
nvme-ioctl.c:259:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) (payload),
            ^
nvme-ioctl.c: In function ‘nvme_resv_report’:
nvme-ioctl.c:271:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) data,
            ^
nvme-ioctl.c: In function ‘nvme_identify’:
nvme-ioctl.c:295:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) data,
            ^
nvme-ioctl.c: In function ‘nvme_get_log’:
nvme-ioctl.c:328:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) data,
            ^
nvme-ioctl.c: In function ‘nvme_feature’:
nvme-ioctl.c:373:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) data,
            ^
nvme-ioctl.c: In function ‘nvme_ns_create’:
nvme-ioctl.c:428:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) ((void *)&ns),
            ^
nvme-ioctl.c: In function ‘nvme_ns_attachment’:
nvme-ioctl.c:460:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) cntlist,
            ^
nvme-ioctl.c: In function ‘nvme_fw_download’:
nvme-ioctl.c:487:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) data,
            ^
nvme-ioctl.c: In function ‘nvme_sec_send’:
nvme-ioctl.c:511:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) data,
            ^
nvme-ioctl.c: In function ‘nvme_sec_recv’:
nvme-ioctl.c:532:12: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
   .addr  = (__u64) data,
            ^
cc1: all warnings being treated as errors
Makefile:37: recipe for target 'nvme-ioctl.o' failed
make: *** [nvme-ioctl.o] Error 1`

keithbusch, 2016-03-14:
Thank you for the notice, will fix immediately.

keithbusch, 2016-03-14:
Fixed here:

https://github.com/linux-nvme/nvme-cli/commit/43df230e01ea744aaf38f93e6bffdb3798147593

closed 2016-03-14 by keithbusch