linux-nvme/nvme-cli #81: nvme0 failed to map

sonnyp, 2016-04-16:
```
root@kirby /h/sonny# nvme list
nvme0 failed to map
```

Linux 4.5.0
nvme-cli 0.6

keithbusch, 2016-04-18:
Did this work for you on some other version of the kernel or cli?

sonnyp, 2016-04-18:
works on Linux `4.4.7-1-lts`

helge000, 2016-05-02:
same issue with kernel 4.6 branch

keithbusch, 2016-05-02:
puzzled... There are no driver specific changes to exposing pci "resource0", and it seems to work fine for me.

```
# uname -r
4.6.0-rc5+
# nvme list
Node             SN                   Model                                    Version  Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- -------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     CVMD4183000W2P0EGN   INTEL SSDPEDME020T4                      1.0      1           2.00  TB /   2.00  TB    512   B +  0 B   8DAAZRV1
/dev/nvme1n1     CVFT40230035800CGN   INTEL SSDPEDMD800G4                      1.0      1         800.17  GB / 800.17  GB    512   B +  0 B   8DV10184

```

If it works in 4.4, but breaks in 4.5, anyone who can reproduce this care to bisect it?

helge000, 2016-05-02:
AFAICT `list` - and commands depending on it - is the only command witch does not work

keithbusch, 2016-05-02:
How about "nvme show-regs nvme0"? Does that also fail?

helge000, 2016-05-02:
Same, does not work. I should mention I am running `nvme-cli-0.5-1.fc23.x86_64`

helge000, 2016-05-02:
Also, Samsiung?

keithbusch, 2016-05-02:
I didn't think the controller vendor would matter for this particular case (we're just mmap'ing a sysfs resource file), but i suppose stranger things have happened.

Could you run ``strace nvme show-regs nvme0`` and send output?

helge000, 2016-05-02:
```
execve("/sbin/nvme", ["nvme", "show-regs", "nvme0"], [/* 21 vars */]) = 0
brk(NULL)                               = 0x55a809db4000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f60ec1b7000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=131263, ...}) = 0
mmap(NULL, 131263, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f60ec196000
close(3)                                = 0
open("/lib64/libudev.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=133928, ...}) = 0
mmap(NULL, 134624, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f60ec175000
mprotect(0x7f60ec193000, 4096, PROT_NONE) = 0
mmap(0x7f60ec194000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1e000) = 0x7f60ec194000
close(3)                                = 0
open("/lib64/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\240\6\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=2103656, ...}) = 0
mmap(NULL, 3934784, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f60ebbd7000
mprotect(0x7f60ebd8e000, 2097152, PROT_NONE) = 0
mmap(0x7f60ebf8e000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1b7000) = 0x7f60ebf8e000
mmap(0x7f60ebf94000, 14912, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f60ebf94000
close(3)                                = 0
open("/lib64/libselinux.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\260b\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=136600, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f60ec174000
mmap(NULL, 2237248, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f60eb9b4000
mprotect(0x7f60eb9d3000, 2097152, PROT_NONE) = 0
mmap(0x7f60ebbd3000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1f000) = 0x7f60ebbd3000
mmap(0x7f60ebbd5000, 4928, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f60ebbd5000
close(3)                                = 0
open("/lib64/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0`\16\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=19776, ...}) = 0
mmap(NULL, 2109712, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f60eb7b0000
mprotect(0x7f60eb7b3000, 2093056, PROT_NONE) = 0
mmap(0x7f60eb9b2000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f60eb9b2000
close(3)                                = 0
open("/lib64/librt.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0P!\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=43784, ...}) = 0
mmap(NULL, 2128800, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f60eb5a8000
mprotect(0x7f60eb5af000, 2093056, PROT_NONE) = 0
mmap(0x7f60eb7ae000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6000) = 0x7f60eb7ae000
close(3)                                = 0
open("/lib64/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0PU\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1141184, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f60ec173000
mmap(NULL, 3150104, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f60eb2a6000
mprotect(0x7f60eb3a7000, 2093056, PROT_NONE) = 0
mmap(0x7f60eb5a6000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x100000) = 0x7f60eb5a6000
close(3)                                = 0
open("/lib64/libresolv.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\2009\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=114984, ...}) = 0
mmap(NULL, 2206280, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f60eb08b000
mprotect(0x7f60eb0a2000, 2097152, PROT_NONE) = 0
mmap(0x7f60eb2a2000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x17000) = 0x7f60eb2a2000
mmap(0x7f60eb2a4000, 6728, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f60eb2a4000
close(3)                                = 0
open("/lib64/libdw.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\20\227\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=307256, ...}) = 0
mmap(NULL, 2397128, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f60eae41000
mprotect(0x7f60eae88000, 2093056, PROT_NONE) = 0
mmap(0x7f60eb087000, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x46000) = 0x7f60eb087000
close(3)                                = 0
[... 101 lines cut ...]
mprotect(0x55a808106000, 4096, PROT_READ) = 0
mprotect(0x7f60ec1b8000, 4096, PROT_READ) = 0
munmap(0x7f60ec196000, 131263)          = 0
set_tid_address(0x7f60ec16cad0)         = 7179
set_robust_list(0x7f60ec16cae0, 24)     = 0
rt_sigaction(SIGRTMIN, {0x7f60ea80db50, [], SA_RESTORER|SA_SIGINFO, 0x7f60ea8189f0}, NULL, 8) = 0
rt_sigaction(SIGRT_1, {0x7f60ea80dbe0, [], SA_RESTORER|SA_RESTART|SA_SIGINFO, 0x7f60ea8189f0}, NULL, 8) = 0
[... cut ...]
open("/usr/lib/locale/locale-archive", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=111950656, ...}) = 0
mmap(NULL, 111950656, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f60e306a000
close(3)                                = 0
open("/sys/class/nvme/nvme0/device/resource0", O_RDONLY) = 3
mmap(NULL, 4096, PROT_READ, MAP_SHARED, 3, 0) = -1 EINVAL (Invalid argument)
write(2, "nvme0 failed to map\n", 20nvme0 failed to map
)   = 20
exit_group(19)                          = ?
+++ exited with 19 +++
```

keithbusch, 2016-05-02:
Alright, either the memory mapped region is too large (doubtful -- an NVMe bar has to be at least 8k, and we're just mapping 4k), or the BAR is flagged IORESOURCE_EXCLUSIVE, which I've never seen used before. The NVMe driver at least does not use it.

Could you run ``cat /sys/class/nvme/nvme0/device/resource``?

helge000, 2016-05-02:
Certainly:
```
0x00000000f1000000 0x00000000f1003fff 0x0000000000140204
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x000000000000d000 0x000000000000d0ff 0x0000000000040101
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
```

keithbusch, 2016-05-02:
Hm, that looks like pretty standard configuration to me, very similar to my Intel cards. I'm not able to match anything in the code to returning EINVAL with these resource types. It looks like it should work fine. I'll take another look tomorrow.

sonnyp, 2016-05-02:
For me, in case it's helpful
```
sonny@kirby ~> cat /sys/class/nvme/nvme0/device/resource
0x00000000f7300000 0x00000000f7303fff 0x0000000000140204
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x000000000000e000 0x000000000000e0ff 0x0000000000040101
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
```

keithbusch, 2016-05-03:
Are there by any chance any new dmesg kernel prints logged after running the failing command?

Could we also get the output from the following commands (replacing \<B:D.f\> with the appropriate bus-dev-func numbers for your NVMe controller):

```
cat /proc/iomem
lspci -vvv -s <B:D.f>
```

sonnyp, 2016-05-03:
no dmesg message

```
sonny@kirby ~/P/J/pointer> lspci -vvv -s 01:00.0
01:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller (rev 01) (prog-if 02 [NVM Express])
	Subsystem: Samsung Electronics Co Ltd Device a801
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 16
	NUMA node: 0
	Region 0: Memory at f7300000 (64-bit, non-prefetchable) [size=16K]
	Region 2: I/O ports at e000 [size=256]
	Capabilities: <access denied>
	Kernel driver in use: nvme
	Kernel modules: nvme

```

```
sonny@kirby ~/P/J/pointer> cat /proc/iomem
00000000-00000fff : reserved
00001000-00057fff : System RAM
00058000-00058fff : reserved
00059000-0009efff : System RAM
0009f000-0009ffff : reserved
000a0000-000bffff : PCI Bus 0000:00
000c0000-000cfbff : Video ROM
000f0000-000fffff : System ROM
00100000-7d1ee017 : System RAM
  01000000-015b2398 : Kernel code
  015b2399-018f2aff : Kernel data
  01a27000-01b47fff : Kernel bss
7d1ee018-7d1fec57 : System RAM
7d1fec58-7d1ff017 : System RAM
7d1ff018-7d20f057 : System RAM
7d20f058-800e6fff : System RAM
800e7000-800e7fff : ACPI Non-volatile Storage
800e8000-80101fff : reserved
80102000-801a7fff : System RAM
801a8000-80cb8fff : reserved
80cb9000-84c9ffff : System RAM
84ca0000-86366fff : reserved
86367000-8639dfff : ACPI Tables
8639e000-86b61fff : ACPI Non-volatile Storage
86b62000-873fefff : reserved
873ff000-873fffff : System RAM
87400000-87ffffff : RAM buffer
88000000-c7ffffff : reserved
  88000000-c7ffffff : Graphics Stolen Memory
c8000000-f7ffffff : PCI Bus 0000:00
  c8000000-c81fffff : PCI Bus 0000:03
  c8200000-c83fffff : PCI Bus 0000:03
  e0000000-efffffff : 0000:00:02.0
  f6000000-f6ffffff : 0000:00:02.0
  f7000000-f71fffff : PCI Bus 0000:04
    f7000000-f71fffff : 0000:04:00.0
      f7000000-f71fffff : ath
  f7200000-f72fffff : PCI Bus 0000:05
    f7200000-f7207fff : 0000:05:00.0
      f7200000-f7207fff : xhci-hcd
  f7300000-f73fffff : PCI Bus 0000:01
    f7300000-f7303fff : 0000:01:00.0
      f7300000-f7303fff : nvme
  f7400000-f741ffff : 0000:00:1f.6
    f7400000-f741ffff : e1000e
  f7420000-f742ffff : 0000:00:1f.3
    f7420000-f742ffff : ICH HD audio
  f7430000-f743ffff : 0000:00:14.0
    f7430000-f743ffff : xhci-hcd
  f7440000-f7443fff : 0000:00:1f.3
    f7440000-f7443fff : ICH HD audio
  f7444000-f7447fff : 0000:00:1f.2
  f7448000-f7449fff : 0000:00:17.0
    f7448000-f7449fff : ahci
  f744a000-f744a0ff : 0000:00:1f.4
  f744b000-f744b7ff : 0000:00:17.0
    f744b000-f744b7ff : ahci
  f744c000-f744c0ff : 0000:00:17.0
    f744c000-f744c0ff : ahci
  f744d000-f744dfff : 0000:00:16.0
    f744d000-f744dfff : mei_me
  f7fc0000-f7fdffff : pnp 00:05
f8000000-fbffffff : PCI MMCONFIG 0000 [bus 00-3f]
  f8000000-fbffffff : reserved
    f8000000-fbffffff : pnp 00:05
fd000000-fe7fffff : PCI Bus 0000:00
  fd000000-fdabffff : pnp 00:06
  fdac0000-fdacffff : pnp 00:08
  fdad0000-fdadffff : pnp 00:06
  fdae0000-fdaeffff : pnp 00:08
  fdaf0000-fdafffff : pnp 00:08
  fdb00000-fdffffff : pnp 00:06
  fe000000-fe010fff : reserved
  fe036000-fe03bfff : pnp 00:06
  fe03d000-fe3fffff : pnp 00:06
  fe410000-fe7fffff : pnp 00:06
fec00000-fec00fff : reserved
  fec00000-fec003ff : IOAPIC 0
fed00000-fed003ff : HPET 0
  fed00000-fed003ff : PNP0103:00
fed10000-fed17fff : pnp 00:05
fed18000-fed18fff : pnp 00:05
fed19000-fed19fff : pnp 00:05
fed20000-fed3ffff : pnp 00:05
fed45000-fed8ffff : pnp 00:05
fed90000-fed93fff : pnp 00:05
fee00000-fee00fff : Local APIC
  fee00000-fee00fff : reserved
ff000000-ffffffff : reserved
  ff000000-ffffffff : INT0800:00
    ff000000-ffffffff : pnp 00:05
100000000-436ffffff : System RAM
437000000-437ffffff : RAM buffer
```

keithbusch, 2016-05-03:
Let's try to map registers directly instead of relying on pci-sysfs, Could you try to run the following?

```
# wget http://git.infradead.org/users/kbusch/nvme-user.git/blob_plain/a1b0e61972d6dc1d259f06a0d78eb9ad8a3d6918:/nvme_read_regs.c
# gcc -m64 nvme_read_regs.c -o nvme_read_regs
# sudo ./nvme_read_regs 0xf7300000
```

I'm pulling the "0xf7300000" based on your output from resource and lspci, but change accordingly if the address is different for any other system.

sonnyp, 2016-05-03:
`terminated by signal SIGSEGV (Address boundary error)`

keithbusch, 2016-05-03:
Well, I'm totally clueless here. This has nothing to do with the driver, pci core, or nvme-cli.

Can someone who's hitting this bisect 4.4 to 4.5 and see what kernel commit changed this behavior?

helge000, 2016-05-03:
I got this, however:
```                                                         
./nvme_read_regs 0x<bar>
```

keithbusch, 2016-05-03:
That's printing the usage as it detects you didn't provide the correct arguments. What was the exact command you ran?

helge000, 2016-05-03:
Right - missed that - sorry! Here you go:
```
cap     : ffffffffffffffff
version : ffffffff
intms   : ffffffff
intmc   : ffffffff
cc      : ffffffff
csts    : ffffffff
aqa     : ffffffff
asq     : ffffffffffffffff
acq     : ffffffffffffffff
cmbloc  : ffffffffffffffff
cmbsz   : ffffffffffffffff
```

keithbusch, 2016-05-03:
Try using "0xf1000000" on your system instead of 0xf7300000. :)

If you did use 0xf1000000, then that's just weird.

helge000, 2016-05-03:
> try using "0xf1000000" on your system instead of 0xf7300000. :)

Ok, that gets me the segfault, too

helge000, 2016-05-03:
If it helps, the `lspci` output from my nvme conroller.

```
05:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller (rev 01) (prog-if 02 [NVM Express])
	Subsystem: Samsung Electronics Co Ltd Device a801
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 16
	Region 0: Memory at f1000000 (64-bit, non-prefetchable) [size=16K]
	Region 2: I/O ports at d000 [size=256]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable- Count=1/8 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [70] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+
		DevCtl:	Report errors: Correctable+ Non-Fatal+ Fatal+ Unsupported+
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+ FLReset-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ UncorrErr- FatalErr+ UnsuppReq+ AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x4, ASPM L1, Exit Latency L0s <4us, L1 <64us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s, Width x4, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis+, LTR+, OBFF Not Supported
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+, EqualizationPhase1+
			 EqualizationPhase2+, EqualizationPhase3+, LinkEqualizationRequest-
	Capabilities: [b0] MSI-X: Enable+ Count=9 Masked-
		Vector table: BAR=0 offset=00003000
		PBA: BAR=0 offset=00002000
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
	Capabilities: [148 v1] Device Serial Number 00-00-00-00-00-00-00-00
	Capabilities: [158 v1] Power Budgeting <?>
	Capabilities: [168 v1] #19
	Capabilities: [188 v1] Latency Tolerance Reporting
		Max snoop latency: 3145728ns
		Max no snoop latency: 3145728ns
	Capabilities: [190 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=10us PortTPowerOnTime=10us
	Kernel driver in use: nvme
	Kernel modules: nvme
```

keithbusch, 2016-05-03:
This issue is beyond me as this has nothing to do with the nvme-cli or the linux-nvme driver anymore. I don't know much about the memory management code.

If I could repro this on my machine, I'd bisect the kernel ... then write an angry email to the committer.

helge000, 2016-05-03:
Do you habe a starting point? I just randomly build tags v0.3, 0.4, 0.6 and master.
< 0.5 I get a seg fault, >= 0.5 I get  `nvme0 failed to map`

keithbusch, 2016-05-03:
I was talking about the kernel rather than the cli. I hear kernel 4.4 worked, but 4.5 does not. It's roughly 13 steps to bisect.

Are you saying that some older version of cli works in 4.5?

helge000, 2016-05-03:
Ah, ok! Might be kernel... yep; bisecting this is beyond me :)

helge000, 2016-05-03:
> Are you saying that some older version of cli works in 4.5?

I have a Skylake machine and kernels below 4.6 constantly crash; hence I am running currently Fedora Rawhide kernels. I could test it, however. 

In the end, it `only` seems to affect the list function; the rest works for me.

keithbusch, 2016-05-03:
Yeah, the "list" and "show-regs" functions are special. They're the only ones mapping bars into user space. The rest are purely passthrough nvme ioctl commands.

spheenik, 2016-06-12:
Just adding myself to the list, same issue, using 4.6.0-mainline.

keithbusch, 2016-06-13:
Still successful for me. Can anyone confirm which kernel version this was working?

spheenik, 2016-06-13:
I think it was working in 4.4. So it looks like it began with 4.5.

sonnyp, 2016-06-13:
I confirm it failed with 4.5 and worked with 4.4

keithbusch, 2016-06-13:
Ok, for some combination of this device and your platform, the kernel is mapping the resources in 4.4 differently than in 4.5. What it's doing differently is a mystery to me. All externally visible attributes I'm aware of that we can query look correct.

Still think this needs to be bisected with git at the kernel level.

Stephenm64, 2016-06-28:
Just a heads up that i have personally seen and replicated this issue on Centos 7. (nvme 0.7.44.x)

The issue was fixed by disabling secure boot on the system.

keithbusch, 2016-06-29:
Very interesting, thanks for the info.

Not sure we can recommend disabling secure boot just for this program. :)

scarsxxx, 2016-07-07:
I had a same problem on my skylake desktop with kernel 4.4.0-21.

iori-yja, 2016-07-14:
Kernel 4.4.0-21 can not run `nvme list` is curious.
In my environment, `nvme list` can run with 4.4.14-1-lts, however cannot run with 4.6.3-1.

Also, I don't think disabling/enabling secureboot matters. I guess his signed kernel for secureboot is just older than another (maybe 4.4 or 4.5?).

richardschuetz, 2016-08-29:
See https://github.com/torvalds/linux/commit/90a545e981267e917b9d698ce07affd69787db87 for explanation of the changed kernel behavior.

jderrick, 2016-08-29:
> See torvalds/linux@90a545e
> for explanation of the changed kernel behavior.

Does this actually apply to nvme-cli? We mmap the pci resource file, but
not /dev/mem

jderrick, 2016-08-29:
My mistake, just saw the relevant:
"Any application using /dev/mem or mmap of sysfs pci resources will now need to perform the extra step of either:"

iori-yja, 2016-08-29:
With 4.7.2, whether it works or not is depending on kernel configuration.
With a kernel with my own configuration, it can list all devices up, however, with Archlinux default configuration, it cannot.

NVMe-concerning options (`NVM Express block device` and `SCSI emulation for NVMe device node`) are both enabled on each environment.

I haven't figured out which option is exactly affecting, but I wonder why such circumstance happens. If the behavior change is the only reason `nvme list` can run with some kernel and can't run with the others, I have no idea why it can conditionally run with recent kernel....

keithbusch, 2016-08-29:
It's dependent on CONFIG_IO_STRICT_DEVMEM.

We can't support mapping the controller's pci bar under these conditions, but there's no reason to make the 'list' command depend on it either. I'll push a fix this week unless someone beats me to it.

iori-yja, 2016-08-29:
@keithbusch Thank you, it's just the moment I found that as noted in the Linus' explanation.
I'm testing switching on and off only the config and whether it reproduced that.

keithbusch, 2016-08-30:
It took a while, but thanks to @richardschuetz for pointing to the kernel config that caused different results.

I've removed the register mapping dependency for the list command to run. Commit is here: https://github.com/linux-nvme/nvme-cli/commit/6ab2e9ef204053d82c1eb40e514119286749cefe

Closing this issue since commit is added.

closed 2016-08-30 by keithbusch