linux-nvme/nvme-cli #84: NVME Admin command error: INVALID_OPCODE(2001)

mdp2000, 2016-05-09:
I purchased Parted Magic to erase the drive in a new X1 Carbon 20FB-004QUS, however, although I can run Erase Disk - NVMe Secure Erase option and see the drive (Samsung MZVKV512HAJH-000L1), the erase process ends with "/dev/nvme0n1: Error! Disk is not erased". I then attempted a command line "nvme format /dev/nvme0n1 --ses=1", as well as "nvme format /dev/nvme0n1" (which a Parted Magic support forum admin confirmed is the command they use), but in both cases I get "Admin command error:INVALID_OPCODE(2001)" as the result.  The Parted Magic forum admin requested that I contact you directly for assistance.  Any feedback on how to correctly secure erase this drive is appreciated.

Thank you.

keithbusch, 2016-05-09:
That doesn't sound right since format is not an optional command to support for any device. It's required. Perhaps the device's status is incorrect, and maybe it doesn't like using the "all namespaces". Do you get the same error from running:

``# nvme format /dev/nvme0 -n 1``

?

mdp2000, 2016-05-09:
Yes, same error.

keithbusch, 2016-05-09:
Okay, in this case, I recommend forwarding the error to the controller vendor. It sounds like they require something not defined in the specification, or they simply don't support the nvme format admin command.

keithbusch, 2016-05-10:
Closing; this appears to be a device side issue.

closed 2016-05-10 by keithbusch

fuzzydunlop1899, 2017-05-03:
Hi,

I have the exact same problem with the Samsung SM951 256 GB NVMe drive (MZVPV256HDGL-00000) and was also routed here via the Parted Magic forum. Since the SM951 is an unsupported OEM-Product I hope there might be a so workaround by now.

Thank you!

keithbusch, 2017-05-03:
I think the work-around is to just not issue format commands. The device doesn’t support it according to the status code.

eheiman, 2017-07-06:
Samsung PM961 128G responds with NVME Admin command error:8193 when nvme_format_ns /dev/nvme0n1 is sent on ASrock Taichi Z270 running CentOS 7.3.1611.el7.
Same SSD/same command/same OS issued on ASUS Maximus Gene VII Z97 completes successfully. Please advise possible problem(s) on Z270 system.

keithbusch, 2017-07-06:
That looks like you're using an older version of this tool, so I'm going to assume 8193 is in decimal. That's 2001 in hex, and according to the spec, that's an invalid opcode error. Supporting the format opcode is optional, so it sounds like controller chose not to support it, and that's okay from a specification standpoint.

eheiman, 2017-07-06:
Thank you for this reply. Using the very latest version of the tool gives this response:
[root@cabench11 Desktop]# nvme format /dev/nvme0n1
NVME Admin command error:INVALID_OPCODE(2001)

Issuing this command to this SSD on ASUS Z97 with same OS completes successfully. Same behavior with other Samsung SM/PM version SSDs???

Other thoughts?

keithbusch, 2017-07-06:
If you want to verify if your controller supports the Format NVM command, run `nvme id-ctrl /dev/nvme0 | grep oacs`. If value does not have bit 1 set, your controller doesn't support the command, and there's nothing wrong with that from an implementation compliance standpoint.

eheiman, 2017-07-06:
Both Z270 and Z97 systems yield oacs    : 0x7. I suspect OP_CODE is modified by Z270 versus Z97. Comments will be appreciated.
[Samsung_format_Z270.txt](https://github.com/linux-nvme/nvme-cli/files/1129268/Samsung_format_Z270.txt)

keithbusch, 2017-07-06:
An oacs of 7 is advertising support for format nvm. If it's not working, you should take this issue to your vendor.

cbeyls, 2017-08-04:
I owned 2 laptops with Samsung PM951 and PM961 SSDs and both had the same issue: oacs shows support but the format command returns INVALID_OPCODE(2001).
For both machines, **the fix is to put to laptop to sleep**, then resume it. After resume, the Samsung SSD accepts the command.

eheiman, 2017-08-07:
Your comment is most appreciated. This same remedy also works for a SATA SSD in security frozen state. Based on that observation, I suspect this is a security related issue, present on Z270 but not on Z97. Review of BIOS configurations is in order.

fallspectrum, 2017-08-09:
Just confirming that putting the machine to sleep has also resolved this issue for me.

PeterSurda, 2017-08-26:
I know it's closed but I'm just adding that I had the same problem and the same solution (suspend + resume) worked. Laptop with Intel HM175 Express Chipset and Samsung SM961 512GB.

keithbusch, 2017-08-28:
That sounds like a firmware bug.

In any case, I hope everyone running this command is aware it obliterates your data. :)

toyg, 2018-05-24:
Almost a year later, and Samsung NVMEs are still behaving in the same way.

```
$> sudo nvme format /dev/nvme0n1
NVME Admin command error:INVALID_OPCODE(2001)
$> systemctl suspend
$> sudo nvme format /dev/nvme0n1
Success formatting namespace:1
```

trourance, 2018-10-12:
It's not working for me if I boot on an usb key to format it.

```
[liveuser@localhost ~]$ sudo nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     S308NAAH500323       SAMSUNG MZSLW1T0HMLH-000L1               1         220.14  GB /   1.02  TB    512   B +  0 B   3L0QCXY7

[liveuser@localhost ~]$ sudo nvme format -s1 /dev/nvme0n1
NVME Admin command error:INVALID_OPCODE(2001)
[liveuser@localhost ~]$ systemctl suspend 
[liveuser@localhost ~]$ sudo nvme format -s1 /dev/nvme0n1
NVME Admin command error:INVALID_OPCODE(2001)
```

KaLisdef, 2018-11-13:
Hi everyone,
I've just tested the nvme format command and it worked with a live debian/gnome on a Samsung 970 EVO.
I did not had to enter in a sleep state.

nama08, 2019-06-04:
Hi,
I am having challenges with NVME format on Samsung drive **(MZVLB256HAHQ).**

**Error:** NVME Admin command error:INVALID_OPCODE(2001)

**Note:** I have been contacting Samsung support, and they weren't able to help because this particular NVMe drive was OEM. When contacted Lenovo /OEM, but no luck there as well as they do not provide support for these drives.

cbeyls, 2019-06-04:
@nama08 To perform a low-level format on Lenovo Thinkpad machines (both SATA and NVME), you have to use the Lenovo EFI application which allows to reset the cryptographic key and erase the drive: https://support.lenovo.com/be/fr/downloads/ds019026
This tool works with all generations of Thinkpad machines including the most recent ones.

nama08, 2019-06-04:
> @nama08 To perform a low-level format on Lenovo Thinkpad machines (both SATA and NVME), you have to use the Lenovo EFI application which allows to reset the cryptographic key and erase the drive: [...]

Hi and thanks for your valuable reply. I will check this out. As mentioned earlier, I was in contact with Samsung support and they weren't able to help. 
By the way is this process is NIST compliant?
Support for Linux based environment?

cbeyls, 2019-06-04:
You need to have a Thinkpad machine listed on the Lenovo tool page. I personally own a T480S with a Samsung NVME drive (model MZVLB256) and I can confirm the tool works with this drive (and I could not use `nvme format` with it either). The Lenovo tool is a bootable USB key image (UEFI application) so the installed OS is irrelevant. After you use the tool, your drive will be wiped clean and you can then install any OS, including Linux.

I was also able to upgrade the firmware of the Samsung drive directly from Linux using the `nvme` command line tool, by identifying the proper file in the [Windows zip archive](https://pcsupport.lenovo.com/be/fr/downloads/ds119265) and following [these instructions](https://gist.github.com/klingtnet/22ab0b907e2d9d20f98c72c93ea5dd37).

nama08, 2019-06-05:
Hello again, I will check it out if I can run the actual utility under Linux.

comicfans, 2020-01-14:
I have wd sn 750 1TB, nvme format gives error first,(change block size from 512 to 4096, got error code 0x4001, I'm not sure). then I tried supend/resume and it helped me

aggieNick02, 2022-01-14:
Same drive as @comicfans (model number WDBRPG0010BNC-WRSN). On a brand new never touched drive, just running
`sudo nvme format /dev/nvme0n1`
returned
`NVMe status: INVALID_FORMAT: The LBA Format specified is not supported. This may be due to various conditions(0x410a)`

Suspend and resume fixed the issue, with the original command running succesfully after resume. So bizarre.

Lantizia, 2024-02-12:
Just so @trourance doesn't feel he's not the only one.

I have a Dell XPS 9360 which came with this Samsung PM961 NVMe 512GB drive...
[image]
P/N: MZVLW512HMJP - 000D1
Model: MZ - VLW512A
DP/N: 068V6F

Using the latest nvme-cli version 2.7.1 via the AppImage (and I tried compiling manually too) I get this...

(this area will be edited momentarily with the output)

Does anyone know of any other workaround?  

Also is CXY74D1Q the most recent known (and Dell) firmware for this SSD?  (fwupd / LVFS doesn't seem to know of anything later it seems)

Lantizia, 2024-02-12:
In the end I've found that the Dell BIOS supports a 'data wipe' and it's managed to issue the command instead... the SSD was wiped in only a handful of seconds.

But then that does mean there isn't a problem with the device... so the issue must be how nvme-cli communicates with it?

Should this issue be reopened?

keithbusch, 2024-02-12:
The Dell BIOS is more than likely performing some kind of vendor unique sequence to unlock this operation. There's nothing wrong with the nvme-cli tooling from the protocol perspective, but we don't inherently know vendor specific magic. We'd need to know what command sequence the special BIOS updater is doing in order to script up an equivalent with this tool.

aggieNick02, 2024-06-21:
I recently had to revisit this issue on a PC where the suspend workaround didn't work, and now have a pretty full understanding of what the real issue is, why the workaround sometimes works, and what can be done when it doesn't. 

### Quick Summary

- The issue has nothing to do with `nvme-cli` (as expected)
- The issue is caused by storage devices which behave incorrectly when security operations are blocked
  - motherboards generally block storage device security operations on drives at boot
- The suspend/resume workaround often works because many motherboards fail to block storage device security operations on resume
  -  But some motherboards do block these on resume; on such boards, the workaround is ineffective
- If you encounter the issue and the suspend/resume workaround is ineffective, you motherboard likely has an option to disable the storage device security operations block on next boot (on my board, it is `Disable Block SID` under `Security>Trusted Computing`), and this will let you `nvme format`, etc during the next boot

### Full Explanation

Most modern PCs, on boot, will use a [Trusted Computing Group (TCG) feature called Block SID Authentication](https://trustedcomputinggroup.org/wp-content/uploads/TCG_Storage-Feature_Set_Block_SID_Authentication_v1.01_r1.00.pdf) on storage devices that support it. For storage devices that support it, this feature will check to see if the storage device's security identifier (SID) password is still set to the default. If it is, the feature will be enabled, and the storage device will block attempts to authenticate until a subsequent device power cycle occurs. This functionality is similar to the ATA FreezeLock command. The goal of this feature is to prevent malware from being able to easily set the password on a disk that has not yet had a password set.

In general, an NVMe device can still be formatted even when SID Authentication is blocked. However, some devices like the WD SN750 (with certain firmware versions) mistakenly block `nvme format` when SID Authentication is blocked. I imagine the other drives mentioned here behave similarly.

Suspending and resuming the system is an effective workaround in many cases because many systems fail to block SID Authentication when resuming from a suspend. As such, doing a suspend and resume disables the block and avoids the arguably buggy behavior of the SN750 and other drives.

If your motherboard blocks SID Authentication when resuming, the workaround is not effective. What can you do? Most boards have a mechanism to allow you to disable the block on next boot - on mine it is `Disable Block SID` under `Security>Trusted Computing`, and only shows up if a drive with blocked SID Authentication is present; You enable `Disable Block SID`, save and exit BIOS, and the system reboots. It then requires confirmation via specific keyboard button press on the next boot that you really want to boot without blocking SID Authentication. Then you can perform the format, or you can set the SID password to a non-default value, which will prevent the SID Authentication block from ever being set again on the storage device.

Examining the Block SID Authentication state of a drive and changing its password is pretty awkward. The SN750 seems to implement TCG Pyrite, or something like it, and the tools for working with such drives are limited. There is a tool called `sedutil-cli`, but the official version of it on Github does not support Pyrite, and no version appears to be available via `apt` in Ubuntu. Fortunately, there are a couple forks that have added support for Pyrite, including one for BSD and one from the owners of sedutil.com. The latter is available on Github with source and precompiled linux/windows binaries: https://github.com/ChubbyAnt/sedutil/releases

After building or copying the prebuilt sedutil-cli, one can examine the state of their drives from within the OS:

```
:~$ sudo ./sedutil-cli --scan
Scanning for Opal compliant disks
/dev/nvme0     p   WDS100T3X0C-00SJG0                       102000WD
No more disks present ending scan
```

The `p` tells us the drive supports pyrite. We can then query the drive:

```
~$ sudo ./sedutil-cli --query /dev/nvme0

/dev/nvme0 NVMe WDS100T3X0C-00SJG0                       102000WD <serial_number>
TPer function (0x0001)
    ACKNAK = N, ASYNC = N. BufferManagement = N, comIDManagement  = N, Streaming = Y, SYNC = Y
Locking function (0x0002)
    Locked = N, LockingEnabled = N, LockingSupported = Y, MBRDone = N, MBREnabled = N, MBRAbsent = Y, MediaEncrypt = N
Block SID Authentication function (0x0402)
    SID Blocked State = Y, SID Value State = N, Hardware Reset = Y
Pyrite 1.0 function (0x0302)
    Base comID = 0x7ffe, comIDs = 1, Initial PIN = 0x0, Reverted PIN = 0x0
```

Notice the `SID Blocked State` is `Y`. After disabling the block (either by using the option in the BIOS or the suspend/resume workaround (if it works)), we see this:

```
~$ sudo ./sedutil-cli --query /dev/nvme0

/dev/nvme0 NVMe WDS100T3X0C-00SJG0                       102000WD <serial_number>
TPer function (0x0001)
    ACKNAK = N, ASYNC = N. BufferManagement = N, comIDManagement  = N, Streaming = Y, SYNC = Y
Locking function (0x0002)
    Locked = N, LockingEnabled = N, LockingSupported = Y, MBRDone = N, MBREnabled = N, MBRAbsent = Y, MediaEncrypt = N
Block SID Authentication function (0x0402)
    SID Blocked State = N, SID Value State = N, Hardware Reset = N
Pyrite 1.0 function (0x0302)
[... cut ...]
After running this, you should be able to format the storage device without any workaround in any machine.

The [TCG Storage Opal Integration Guidelines](https://trustedcomputinggroup.org/wp-content/uploads/TCG_Storage_ReferenceDocument_Opal_Integration_Guidelines_v1.00_r1.00.pdf) have even more details on all of this.

Finally, the suspend/resume workaround is a common mechanism to be able to bypass a similar "FreezeLock" on ATA devices. Interestingly, that mechanism works just fine on the motherboard I'm using, even though the motherboard stops it from working on NVMe devices.

keithbusch, 2024-06-21:
> I recently had to revisit this issue on a PC where the suspend workaround didn't work...

Amazing, thank you for writing this up!

HaroPanosyan, 2024-06-21:
Amazing, Thank you.

Lantizia, 2024-06-22:
Well for any owners of Dell systems out there, this seem to be the trick for disabling the block on SID authentication on them...

[https://www.dell.com/support/kbdoc/en-uk/000126083/pre-boot-authentication-will-not-activate-due-to-sed-block-sid-authentication-enabled](https://www.dell.com/support/kbdoc/en-uk/000126083/pre-boot-authentication-will-not-activate-due-to-sed-block-sid-authentication-enabled)

Although at least in my case the XPS 9360 doesn't have this option in its BIOS (and it's on the latest version).  But like I said previously, they've at least given us a built in option to do this in the BIOS under the menu/screen... Maintenance | Data Wipe

Great to finally understand what the real issue is though, thanks @aggieNick02

igaw, 2024-06-24:
I think we should add this info to our documentation, as it seems a very common thing to happen.

BTW, nvme has a very minimalist SED plugin. Not sure if it supports the steps necessary feature this workaround requires though.

aggieNick02, 2024-06-24:
> I think we should add this info to our documentation, as it seems a very common thing to happen.
> 
> BTW, nvme has a very minimalist SED plugin. Not sure if it supports the steps necessary feature this workaround requires though.

Oh nice, thanks for the info. I'll see if I can test it; it looks like it has the necessary commands, I don't know if it works with drives with Pyrite or Opalite, but regardless having support for this appearing in nvme-cli is encouraging since where to find a sedutil with needed fixes is not super obvious.

aggieNick02, 2024-06-24:
> Oh nice, thanks for the info. I'll see if I can test it; it looks like it has the necessary commands, I don't know if it works with drives with Pyrite or Opalite, but regardless having support for [...]

I grabbed Ubuntu 24.04 LTS and quickly tried with the SN750. Unfortunately, it looks like it does not work with Pyrite drives yet. I get an `ioctl IOC_OPAL_DISCOVERY failed` error message. It looks like Greg Joyce (who added the nvme-cli sed plugin) added the ioctl call to Linux to get the basic information out, and my best guess is that would need to be tweaked before it could work on Pyrite and possibly Opalite.

I've no idea how common Pyrite or Opalite drives are, it just is what the problematic SN750 I was dealing with implemented.

igaw, 2024-06-25:
@gjoyce-ibm Maybe Greg has some insights here. I really have no clue when it comes to SED :)

gjoyce-ibm, 2024-06-26:
When I started looking at SED, there were two viable CLIs, sedcli and sedutil. Both projects however went dormant and had no support or new development.

Given that state I decided that nvme-cli was the best course. But given no real requirements I chose to implement only a base set of OPALv2 functionality. This subset allows for many use cases but certainly not all. The implementation relies entirely on the ioctl's provided by the kernel block SED Opal driver (block/sed_opal.c).

I'm not sure how Pyrite support works but it may be that sedcli/sedutil use the NVMe pass-through mechanism to access the drives directly without using the block SED Opal driver.

I'm a little surprised that ioctl IOC_OPAL_DISCOVERY just failed. Is it possible that your kernel doesn't have that patch?

aggieNick02, 2024-06-28:
> I'm a little surprised that ioctl IOC_OPAL_DISCOVERY just failed. Is it possible that your kernel doesn't have that patch?

I was using Ubuntu 24.04LTS. Do you know offhand which kernel version was the first to have the patch? I can try to dig more later to find out if needed.

This is the commit to the fork of sedutil that added Pyrite support in case its helpful:
https://github.com/ChubbyAnt/sedutil/commit/5ad84d84a1afb5c332ce9931286baad25d64b42a

I'll be unable for the next week or two to look at the tech details of this, but am happy to look more and help after that. Thanks for sharing this info and all of the work on adding support for this to nvme-cli. :)

gjoyce-ibm, 2024-06-28:
It went into kernel 6.7.

commit 9fb10726ecc5145550180aec4fd0adf0a7b1d634
Author: Greg Joyce <gjoyce@linux.vnet.ibm.com>
Date:   Fri Jul 21 16:15:32 2023 -0500

    block: sed-opal: Implement IOC_OPAL_DISCOVERY
    
    Add IOC_OPAL_DISCOVERY ioctl to return raw discovery data to a SED Opal
    application. This allows the application to display drive capabilities
    and state.
    
    Signed-off-by: Greg Joyce <gjoyce@linux.vnet.ibm.com>
    Reviewed-by: Christoph Hellwig <hch@lst.de>
    Reviewed-by: Jonathan Derrick <jonathan.derrick@linux.dev>
    Acked-by: Jarkko Sakkinen <jarkko@kernel.org>
    Link: https://lore.kernel.org/r/20230721211534.3437070-2-gjoyce@linux.vnet.ibm.com
    Signed-off-by: Jens Axboe <axboe@kernel.dk>

aggieNick02, 2024-07-01:
Thanks for the info. Ubuntu 24.04LTS is on the 6.8 kernel, so it should have your patch. When I have time in a couple weeks I'll try to dig a bit and get some more info about why it isn't working on the Pyrite device I tried.

gjoyce-ibm, 2024-07-01:
One other possibility... you need the following kernel configuration:

CONFIG_BLK_SED_OPAL=y

aggieNick02, 2024-07-15:
I finally got back and had some time to check this. I'm running Ubuntu 24.04LTS server live (from the USB without installing). Entering the shell from the installer, I do
```
root@ubuntu-server:/# cat /boot/config-6.8.0-31-generic | grep BLK_SED
CONFIG_BLK_SED_OPAL=y
```
so it seems like everything should be in place, at least for proper Opal.

aggieNick02, 2024-07-16:
So I encountered some Opal 2.0 SSDs recently (Crucial P5 1TB and 2TB) that presented with the same issue of being unable to format due to the Block SID feature. Using `sedutil-cli` to do initial setup remedied that issue but introduced a new one.

The output running `--initialSetup` is slightly different with an Opal drive:
```
~$ sudo ./sedutil-cli --initialSetup my_secret_password /dev/nvme0
SID password changed
takeOwnership complete
Locking SP Activate Complete
LockingRange0 disabled
LockingRange0 set to RW
MBRDone set on
MBREnable set on
Initial setup of TPer complete on /dev/nvme0
```
The `MBRDone` and `MBREnable` lines are new. On cold boot, `MBRDone` is set to off and the "shadow MBR" is enabled because `MBREnable` is set to `on`. This feature makes a different portion of the SSD act as the first 128MB of the drive, and this portion is readonly. This causes some simple `fio` commands to fail:

```
~$ fio --name=foo --iodepth=1 --rw=write --bs=128k --size=10M --filename=/dev/nvme0n1 --direct=1
foo: (g=0): rw=write, bs=(R) 128KiB-128KiB, (W) 128KiB-128KiB, (T) 128KiB-128KiB, ioengine=psync, iodepth=1
fio-3.36
Starting 1 process
fio: io_u error on file /dev/nvme0n1: No data available: write offset=0, buflen=131072
fio: pid=1474, err=61/file:io_u.c:1896, func=io_u error, error=No data available
```

Oddly, removing `--direct=1` allows the fio run to succeed, which I'm at a loss to explain. If anyone has a guess as to how that is possible, I'm very interested.

Anyways, the "fix" if you do not care about using the shadow MBR feature is to disable it permanently by running:

`~$ sudo ./sedutil-cli --setMBREnable off my_secret_password /dev/nvme0`

I'll update my original post shortly to reflect this.

aggieNick02, 2024-07-16:
> One other possibility... you need the following kernel configuration:
> 
> CONFIG_BLK_SED_OPAL=y

Update for @gjoyce-ibm and others on the `nvme sed` plugin and Opal/Pyrite:

After running into some Opal 2.0 devices recently, I booted Ubuntu 24.04LTS with one present, and was able to successfully run
`nvme sed discover /dev/nvme0n1`  . Then to be certain, I powered off, switched to a Pyrite device, and booted up, and the same command returned the `ioctl IOC_OPAL_DISCOVERY failed` message.

So it appears that full Opal 2.0 devices work with the `nvme sed` plugin, but Pyrite devices do not.