linux-nvme/nvme-cli #86: nvme-create-ns.txt Has conflicting info between synopsis and options

TaylorSanchez, 2016-05-16:
Is the shorthand for --nsze supposed to be -n or -s?

Synopsis
'nvme create-ns' <device> [--nsze=<nsze> | -n <nsze>]

Options
-s::
--nsze::
https://guides.github.com/features/mastering-markdown/
	The namespace size.

It appears -s is the correct one, when looking at nvme.c:
https://github.com/linux-nvme/nvme-cli/blob/master/nvme.c :
		{"nsze",  's', "NUM", CFG_LONG_SUFFIX, &cfg.nsze,  required_argument, nsze},

keithbusch, 2016-05-16:
It's -s; the man page is wrong. The '-n' option commonly refers to "namespace" in the rest of the possible sub-arguments, so used 's' for size in this command.

Thanks for the report, will fix shortly.

keithbusch, 2016-05-16:
Fixed: https://github.com/linux-nvme/nvme-cli/commit/402a87b610e5d3e3a9c4700eec077eede13d7b64

closed 2016-05-16 by keithbusch