linux-nvme/nvme-cli #97: src/argconfig.c: 3 * poor error checking ?

dcb314, 2016-07-04:
1.

src/argconfig.c:243]: (style) Checking if unsigned variable 'tmp' is less than zero.

Source code is

            uint8_t tmp = strtol(optarg, &endptr, 0);
            if (errno || tmp < 0 || optarg == endptr) {

Was strtoul intended ?

2.

src/argconfig.c:252]: (style) Checking if unsigned variable 'tmp' is less than zero.
src/argconfig.c:261]: (style) Checking if unsigned variable 'tmp' is less than zero.

Duplicates.

keithbusch, 2016-08-24:
Sorry, I totally missed seeing this issue. Yes, this error checking can be improved as you've described with strtoul. Will get that added soon.

keithbusch, 2016-10-10:
committed here: https://github.com/linux-nvme/nvme-cli/commit/d827ed3007a5a9f8376655766067357947da1c75

closed 2016-10-10 by keithbusch