linux-nvme/nvme-cli #104: nvme fw-log not working in version 0.8 - INVALID_FIELD(4002)

garygeike, 2016-08-05:
Using version 0.7 I can issue nvme fw-log command. but version 0.8 fails for same command.

[root@tom400 nvme-cli-0.7]# nvme fw-log /dev/nvme4
Firmware Log for device:nvme4
afi  : 0x44
frs1 : 0x4b4d495050313037 (KMIPP107)
frs2 : 0x4b4d495050313037 (KMIPP107)
frs3 : 0x4b4d495050313037 (KMIPP107)
frs4 : 0x4b51495050313131 (KQIPP111)

[root@tom400 nvme-cli-master.8]# nvme fw-log /dev/nvme4
NVMe Status:INVALID_FIELD(4002)
[root@tom400 nvme-cli-master.8]#

System is IBM Power 8 
Linux tom400 3.10.0-327.22.2.el7.ppc64 #1 SMP Thu Jun 9 10:31:21 EDT 2016 ppc64 ppc64 ppc64 GNU/Linux
[root@tom400 nvme-cli-master.8]# cat /etc/*release
NAME="Red Hat Enterprise Linux Server"
VERSION="7.2 (Maipo)"
ID="rhel"
ID_LIKE="fedora"
VERSION_ID="7.2"
PRETTY_NAME="Red Hat Enterprise Linux"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:redhat:enterprise_linux:7.2:GA:server"
HOME_URL="https://www.redhat.com/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Red Hat Enterprise Linux 7"
REDHAT_BUGZILLA_PRODUCT_VERSION=7.2
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="7.2"
Red Hat Enterprise Linux Server release 7.2 (Maipo)
Red Hat Enterprise Linux Server release 7.2 (Maipo)

keithbusch, 2016-08-05:
Is this PPC little or big endian?

keithbusch, 2016-08-05:
I assume big since i don't see any problems on little endian machines, and the code looks wrong for big endian anyway. Could you try a patch to confirm? I might just push this anyway, but want to be sure if fixes the issue.

```
---
diff --git a/nvme-ioctl.c b/nvme-ioctl.c
index 2a26ca2..c0d659b 100644
--- a/nvme-ioctl.c
+++ b/nvme-ioctl.c
@@ -363,10 +363,8 @@ int nvme_get_log(int fd, __u32 nsid, __u8 log_id, __u32 data_len, void *data)
 		.data_len	= data_len,
 	};
 	__u32 numd = (data_len >> 2) - 1;
-	__le16 numdu, numdl;
+	__u16 numdu = numd >> 16, numdl = numd & 0xffff;
 
-	numdu = htole16((numd & 0xFFFF0000) >> 16);
-	numdl = htole16(numd & 0x0000FFFF);
 	cmd.cdw10 = log_id | (numdl << 16);
 	cmd.cdw11 = numdu;
 --
```

garygeike, 2016-08-05:
I would be happy to try a patch

I added the following to the nvme_get_log function
__u16 numdu = numd >> 16, numdl = numd & 0xffff;
And commented out  these lines.

__le16 numdu, numdl;

numdu = htole16((numd & 0xFFFF0000) >> 16);

numdl = htole16(numd & 0x0000FFFF);

recompiled and now fw-log works … Thank you!

Thank you
Gary

Western Digital Corporation (and its subsidiaries) E-mail Confidentiality Notice & Disclaimer:

This e-mail and any files transmitted with it may contain confidential or legally privileged information of WDC and/or its affiliates, and are intended solely for the use of the individual or entity to which they are addressed. If you are not the intended recipient, any disclosure, copying, distribution or any action taken or omitted to be taken in reliance on it, is prohibited. If you have received this e-mail in error, please notify the sender immediately and delete the e-mail in its entirety from your system.

keithbusch, 2016-08-08:
Thanks, fixed pushed here:

https://github.com/linux-nvme/nvme-cli/commit/1276f057440de52df2aa47be5fb0b4712a0b6646

closed 2016-08-08 by keithbusch