linux-nvme/nvme-cli #106: Feature Request: Decoding of Error Log Entry Hex Codes

zetasyanthis, 2016-08-18:
Not sure if this is in the planned list already, but decoding of the hex codes (at least the non-vendor-specific ones) would be very helpful.  As is, for those not familiar with the spec (myself included), decoding them manually is very difficult.

(When testing NVMe drives in an assembled system, it is desirable to understand *why* the error counter is incrementing, not just that it is.)

hellotwozeroonethree, 2016-08-19:
Hi,

If I want to know more about nvme, Does there exists any documents except the specification?

Thanks

keithbusch, 2016-08-22:
Sure, we can add additional decoding for spec defined errors. Will put it on my todo list.

keithbusch, 2016-08-26:
@zetasyanthis : just to confirm what you're looking for, is it just that you want the "status field" decoded? Here's an example before and after adding the decoding:

Before:
```
 Entry[ 0]
.................
error_count  : 304
sqid         : 16
cmdid        : 0
status_field : 0x4502
parm_err_loc : 0
lba          : 0x60
nsid         : 0x1
vs           : 0
```

After:
```
 Entry[ 0]
.................
error_count  : 304
sqid         : 16
cmdid        : 0
status_field : 0x4502(READ_ERROR)
parm_err_loc : 0
lba          : 0x60
nsid         : 0x1
vs           : 0
```

zetasyanthis, 2016-08-27:
That's perfect. It'd be nice to have the command ID too, if possible.

(And thank you so much. Is there a place I can pop in a few bucks as a tip?)

keithbusch, 2016-08-29:
Great, thanks. I pushed the decoding here: https://github.com/linux-nvme/nvme-cli/commit/767815333295fd542e2565f7a47b611805ea2a4f

For the command ID, that is really just an arbitrarily host assigned cookie. We can't really provide additional decoding for that, and even knowing it is probably not terribly useful outside carefully controlled environments.

Please let us know of any other features or issues. This project only wants to humbly promote NVMe adoption, and its usage is the only compensation I can accept. :)

closed 2016-08-29 by keithbusch