linux-nvme/nvme-cli #109: Latest nvme-cli does not compile (undefined reference to udev...)

samiWaheed, 2016-08-24:
Latest nvme-cli fails with make errors (Ubuntu 14.04 Kernel 3.19)

**make output**
NVME_VERSION = 0.8.35.g8aa3
cc -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.8.35.g8aa3"' -c argconfig.c
cc -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.8.35.g8aa3"' -c suffix.c
cc -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.8.35.g8aa3"' -c parser.c
cc -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.8.35.g8aa3"' -c nvme-print.c
cc -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.8.35.g8aa3"' -c nvme-ioctl.c
cc -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.8.35.g8aa3"' -c nvme-lightnvm.c
cc -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.8.35.g8aa3"' -c fabrics.c
cc -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.8.35.g8aa3"' -c json.c
cc -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.8.35.g8aa3"' -c plugin.c
cc -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.8.35.g8aa3"' -c intel-nvme.c
cc -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.8.35.g8aa3"' -c lnvm-nvme.c
cc -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.8.35.g8aa3"' -c memblaze-nvme.c
cc -D_GNU_SOURCE -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"0.8.35.g8aa3"' nvme.c -ludev -o nvme argconfig.o suffix.o parser.o nvme-print.o nvme-ioctl.o nvme-lightnvm.o fabrics.o json.o plugin.o intel-nvme.o lnvm-nvme.o memblaze-nvme.o
fabrics.o: In function `disconnect_by_nqn':
/tmp/nvme-cli/fabrics.c:779: undefined reference to `udev_new'
/tmp/nvme-cli/fabrics.c:786: undefined reference to `udev_enumerate_new'
/tmp/nvme-cli/fabrics.c:792: undefined reference to `udev_enumerate_add_match_subsystem'
/tmp/nvme-cli/fabrics.c:793: undefined reference to `udev_enumerate_scan_devices'
fabrics.o: In function `disconnect_subsys':
/tmp/nvme-cli/fabrics.c:742: undefined reference to `udev_enumerate_get_list_entry'
/tmp/nvme-cli/fabrics.c:745: undefined reference to `udev_list_entry_get_name'
/tmp/nvme-cli/fabrics.c:745: undefined reference to `udev_enumerate_get_udev'
/tmp/nvme-cli/fabrics.c:745: undefined reference to `udev_device_new_from_syspath'
/tmp/nvme-cli/fabrics.c:748: undefined reference to `udev_device_get_sysattr_value'
/tmp/nvme-cli/fabrics.c:761: undefined reference to `udev_device_unref'
/tmp/nvme-cli/fabrics.c:742: undefined reference to `udev_list_entry_get_next'
fabrics.o: In function `disconnect_by_nqn':
/tmp/nvme-cli/fabrics.c:795: undefined reference to `udev_enumerate_unref'
/tmp/nvme-cli/fabrics.c:798: undefined reference to `udev_unref'
/tmp/nvme-cli/fabrics.c:798: undefined reference to `udev_unref'
/tmp/nvme-cli/fabrics.c:750: undefined reference to `udev_device_get_syspath'
/tmp/nvme-cli/fabrics.c:756: undefined reference to `udev_device_unref'
/tmp/nvme-cli/fabrics.c:795: undefined reference to `udev_enumerate_unref'
/tmp/nvme-cli/fabrics.c:798: undefined reference to `udev_unref'
/tmp/nvme-cli/fabrics.c:753: undefined reference to `udev_device_unref'
collect2: error: ld returned 1 exit status
make: *** [nvme] Error 1

keithbusch, 2016-08-24:
¿!The build detected libudev exists, it didn't throw an error on including libudev.h, but it can't find libudev symbols?!

I'm not getting errors on debian, fedora, or centos. Do you know which previous version didn't fail?

samiWaheed, 2016-08-24:
I have this revision loaded on this system  (nvme-0.8.33.g7eb4.dirty)

keithbusch, 2016-08-24:
Okay, so only the last commit broke this. I can't imagine how, that's not even touching the fabrics code. If you do a ```git revert 91b189679```, does it build for you?

jderrick, 2016-08-24:
Move -ludev to the end of the linking command so it resolves undefined
symbols with ludev last

jderrick, 2016-08-24:
Eg,

diff --git a/Makefile b/Makefile
index dbcce3e..39597dc 100644
--- a/Makefile
+++ b/Makefile
@@ -35,7 +35,7 @@ OBJS := argconfig.o suffix.o parser.o nvme-print.o
nvme-ioctl.o \
        lnvm-nvme.o memblaze-nvme.o

 nvme: nvme.c nvme.h $(OBJS) NVME-VERSION-FILE
-       $(CC) $(CPPFLAGS) $(CFLAGS) nvme.c $(LDFLAGS) -o $(NVME) $(OBJS)
+       $(CC) $(CPPFLAGS) $(CFLAGS) nvme.c -o $(NVME) $(OBJS) $(LDFLAGS)

 nvme.o: nvme.c nvme.h nvme-print.h nvme-ioctl.h argconfig.h suffix.h
nvme-lightnvm.h fabrics.h
        $(CC) $(CPPFLAGS) $(CFLAGS) -c $<

keithbusch, 2016-08-24:
Okay, this must be a linking problem, but I didn't think gcc cared which order you write the arguments. 

It looks like removing the ```#include <udev.h>``` from nvme.c is what's breaking Sam's build.

The pattern matching rule looks like it's missing the LDFLAGS. I think the below should fix that. though it works on my udev systems without this change, so maybe I have a more forgiving environment.

```
diff --git a/Makefile b/Makefile
index dbcce3e..6f57ae9 100644
--- a/Makefile
+++ b/Makefile
@@ -41,7 +41,7 @@ nvme.o: nvme.c nvme.h nvme-print.h nvme-ioctl.h argconfig.h suffix.h nvme-lightn
 	$(CC) $(CPPFLAGS) $(CFLAGS) -c $<
 
 %.o: %.c %.h nvme.h linux/nvme_ioctl.h
-	$(CC) $(CPPFLAGS) $(CFLAGS) -c $<
+	$(CC) $(CPPFLAGS) $(CFLAGS) $(LDFLAGS) -c $<
 
 doc: $(NVME)
 	$(MAKE) -C Documentation
```

samiWaheed, 2016-08-24:
Make after "git revert 91b189679" worked for me.

keithbusch, 2016-08-24:
Okay, the udev header inclusion that was removed in the final build step is what's ruining the build for you. We need fabrics to be able to resolve this without the final linker containing this, though. Do either of the Makefile changes help?

samiWaheed, 2016-08-24:
Change in Makefile from jderrick worked.
Change from Keith still reported errors.

keithbusch, 2016-08-25:
Thanks, will get this updated.

Jon, if you don't mind, I'll add your attribution to the diff you wrote and apply.

jderrick, 2016-08-25:
Thanks Keith!

keithbusch, 2016-08-25:
Fix pushed here: https://github.com/linux-nvme/nvme-cli/commit/09ef228b4ff1f0f92b06dbc83e00e203ae4dd9f1

Sam, thanks for the quick report on the breakage.
Jon, thanks for the quick fix.

closed 2016-08-25 by keithbusch