linux-nvme/nvme-cli #112: Some changes have been made to Reservation commands.

DrYSOH, 2016-09-06:
Hi Keith,

I'm still working on the reservation commands. Regarding NVMe 1.2.1 spec. I've found improper use of codes in your master branch and made some changes as well. In addition, some functions (e.g., resv_*) related to the reservation do not return error code. Please refer to my branch (https://github.com/DrYSOH/nvme-cli/commits/resv_dev). 

Thanks

keithbusch, 2016-09-06:
Thanks, applied.

If you have any more of these where you have the feature working in your own branch, could you please send these as pull requests instead of as an issue?

closed 2016-09-06 by keithbusch

DrYSOH, 2016-09-07:
Okay. Thanks. 

Later, I will make pull requests to this project. if there are some changes.