linux-nvme/nvme-cli #115: Feature Request: NVMe drive locate function

zetasyanthis, 2016-09-15:
The ledctl/ledmon package doesn't seem to have support for NVMe drives yet.  Any chance nvme-cli will be getting that since that project doesn't seem to be moving very quickly?

(I gave the patch [here](https://sourceforge.net/p/ledmon/patches/6/) by @keithbusch a try, but it didn't seem to work on my system. (Could be because I have a PLX PCIe switch in between the motherboard and drives?)

keithbusch, 2016-09-16:
This really depends on your hardware's PCIe slot capabilities. The patch I sent to ledmon/ctl was more a PoC for ones that advertise Attention & Power Indicators Present, and it's inherently unsafe as it subverts the kernel's pciehp slot control command sequence.

For the time being, PCIe SSD LED blinky lights will remain vendor specific until standards define something.

zetasyanthis, 2016-09-16:
Ah, okay. I'll pursue that with the hardware supplier then and revert my rebuilt ledmon RPM. (I have to say I'm surprised that's not part of the NVMe standard, also...)

closed 2016-09-16 by zetasyanthis