linux-nvme/nvme-cli #118: no visible documentation of format of /etc/nvme/discovery.conf

rmriches, 2016-09-22:
There does not appear to be any documentation of the format of file /etc/nvme/discovery.conf, which a new user is apparently required to create.

Details: I installed nvme-cli v0.9 from a ZIP file on a CentOS 7.2 initiator/host machine.  I ran "nvme discover" and got back "No discover params given and no /etc/nvme/discovery.conf conf".  Doing "fgrep -r discovery.conf nvme-cli-0.9" yielded one '#define' line in fabrics.c and notice that binary files fabrics.o and nvme match.  It would be helpful for a new user to have some documentation about the file format.  Reverse engineering the file format from the source code is rather inefficient.

(I'm a new would-be user of the tool.  If somebody could drop me a quick message with a hint or two about the file format, it would be appreciated.)

jpfreyen, 2016-09-22:
On Thu, 2016-09-22 at 08:43 -0700, Robert Riches wrote:

There does not appear to be any documentation of the format of file /etc/nvme/discovery.conf, which a new user is apparently required to create.

Details: I installed nvme-cli v0.9 from a ZIP file on a CentOS 7.2 initiator/host machine. I ran "nvme discover" and got back "No discover params given and no /etc/nvme/discovery.conf conf". Doing "fgrep -r discovery.conf nvme-cli-0.9" yielded one '#define' line in fabrics.c and notice that binary files fabrics.o and nvme match. It would be helpful for a new user to have some documentation about the file format. Reverse engineering the file format from the source code is rather inefficient.

(I'm a new would-be user of the tool. If somebody could drop me a quick message with a hint or two about the file format, it would be appreciated.)

—

There is no discovery.conf.

Please read the README from nvmetcli target tool on a summary of the NVMe Target discovery controller.  The discovery contents are set up on the NVMe Target, not on the host nor with nvme-cli.

In addition, example usage (yes documentation is lacking) of nvme-cli:

nvme discover  -t rdma --traddr=192.168.1.3 --trsvcid=4420 --hostnqn=host1-rogue-nqn

keithbusch, 2016-09-22:
Hm, no one wrote any man pages for the fabrics sub-commands. I'm sure it's on someone's todo list. :)

rmriches, 2016-09-22:
Excellent!  Thank you, jpfreyen, for the example--and all contributors for the tool.

jpfreyen, 2016-09-22:
On Thu, 2016-09-22 at 09:37 -0700, Keith Busch wrote:

Hm, no one wrote any man pages for the fabrics sub-commands. I'm sure it's on someone's todo list. :)

I'll make a plea to recruit people and help (I actually was against having the discover functionality only in nvme-cli, and you can do the same 'connect' via sysfs without nvme-cli, so I just went with the flow on these nvme-cli developments).

igaw, 2022-01-19:
No progess in this bug report for a long time. If you think it's still relevant please reopen.

closed 2022-01-19 by igaw