linux-nvme/nvme-cli #120: Add nvme cmb command

sbates130272, 2016-09-26:
Guys

As CMBs become more popular we want a nvme cmb command to give a more verbose indication of the CMB functionality of a drive. Also, eventually we may have the ability to alter things about the CMB and this function will be a placeholder for doing that through the nvme-cli tool.

I'll work on a PR that (initially) just displays the CMBLOC and CMBSZ values in human consumable form. This is an improvement on nvme show-regs. Comments?

Stephen

keithbusch, 2016-10-05:
There is a potential problem here on newer kernels with CONFIG_IO_STRICT_DEVMEM. The CMB, being a memory remapped register, is only accessible to user land if nothing in the kernel has claimed it if that config option is set, and it looks like most releases are enabling this.

sbates130272, 2016-10-05:
Keith

This is not just a problem for nvme cmb right? This is an issue where userspace can't access any of the MMIO registers in the NVMe controller BAR. Correct?

It seems like this is going to be an issue for a lot of userspace tools that open pcie resources to inspect there contents right?

I see in nvme.c we access the contents of the BAR by opening the relevant sysfs resource file. Are we going to have to do something else now like issue an admin command to get access to the contents of those registers?

BTW a quick grep through all the Kconfig in 4.8 does not find the CONFIG_IO_STRICT_DEVMEM? Where is this CONFIG_ located? I know I have seen it mentioned but I can't for the life of me find it in the source.

Cheers

Stephen

sbates130272, 2016-10-05:
Belay that last comment. Found it. Not sure why my grep missed it.....

sbates130272, 2016-10-05:
Given the repercussions of the new configuration should we open a more broad issue to discuss how we handle this? On idea would be to detect if the relevant CONFIG is set and if so tell the user to either unbound the driver or change the config in their kernel. Both options suck but at least our users won't be confused about what is happening....

jderrick, 2016-10-05:
Yes I have run into a few issues with this config setting already with some
toy programs accessing /dev/mem. Like you said we'll have to use the sysfs
mmap file as our base

igaw, 2022-01-19:
No progess in this bug report for a long time. If you think it's still relevant please reopen.