linux-nvme/nvme-cli #128: nvme cli commands inconsistent after subsystem reset

narasimhan-v, 2016-10-05:
Is this behaviour expected ?

```
# nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     STM0001A8A15         PCIe3 1.6TB NVMe Adapter                 1           1.60  TB /   1.60  TB      4 KiB +  0 B   KMIPP107

# nvme subsystem-reset /dev/nvme0; echo $?
0

# time nvme list; echo $?
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1                          @                                       1           1.60  TB /   1.60  TB      4 KiB +  0 B   p;

real	1m2.110s
user	0m0.001s
sys	0m0.003s
0

# time nvme list; echo $?
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     STM0001A8A15         PCIe3 1.6TB NVMe Adapter                 1           1.60  TB /   1.60  TB      4 KiB +  0 B   KMIPP107

real	0m0.003s
user	0m0.000s
sys	0m0.002s
0

```

keithbusch, 2016-10-05:
Looks like the command timed out, and nvme list isn't looking for negative error codes, so its displaying garbage.

keithbusch, 2016-10-10:
I've pushed a fix to return the error on the condition here: https://github.com/linux-nvme/nvme-cli/commit/7682404019863db5f5b4c176ce77dcd272d8713f

While it will fix printing garbage output, it's not going to fix the timeout. It sounds like your controller is not raising CSTS.NSSRO in accordance with the spec on subsystem reset.

closed 2016-10-10 by keithbusch

narasimhan-v, 2016-10-12:
@keithbusch thanks