linux-nvme/nvme-cli #129: Provide a trap for the  CONFIG_IO_STRICT_DEVMEM issue

sbates130272, 2016-10-07:
Hi

In linux kernel upstream commit 90a545e98 a new CONFIG was added to the kernel that prevents userspace programs (even root) from accesses MMIO regions that are already claimed by a driver. This breaks certain things in NVMe and was found and discussed in another issue.

The issue tracks an effort to document the issue and provide a better user experience when it happens. I plan to try and add a trap for this and a section to the README.md to educate users.

In time we might fix all the broken commands and then this issue becomes moot.

Cheers

Stephen

sbates130272, 2016-10-07:
From <root>/libs/Kconfig.debug

config IO_STRICT_DEVMEM
        bool "Filter I/O access to /dev/mem"
        depends on STRICT_DEVMEM
        ---help---
          If this option is disabled, you allow userspace (root) access to all
          io-memory regardless of whether a driver is actively using that
          range.  Accidental access to this is obviously disastrous, but
          specific access can be used by people debugging kernel drivers.

          If this option is switched on, the /dev/mem file only allows
          userspace access to *idle* io-memory ranges (see /proc/iomem) This
          may break traditional users of /dev/mem (dosemu, legacy X, etc...)
          if the driver using a given range cannot be disabled.

          If in doubt, say Y.

keithbusch, 2016-10-07:
We've got the one command (show-regs) that relies on this being disabled, but it's not a super critical command. Even with the config enabled, it should be possible to use it with a user space driver like SPDK.

closed 2019-03-18 by keithbusch