linux-nvme/nvme-cli #136: Add support for fibre-channel

d-grigsby, 2016-10-24:
Hi Guys,
I would like to add this patch to nvme-cli
[add-fc-support.txt](https://github.com/linux-nvme/nvme-cli/files/548591/add-fc-support.txt)
 tool to add support for a fibre-channel host address as well as a port address. If you are Ok with this change then I'll submit a pull request.:
   nvme-cli: Add support for Fibre-channel host address
    
       Example:
        nvme connect-all --transport=fc --traddr=pn-0x200100110d9f4100:nn-0x200100110d9f4100 \
        --host_traddr=nn-0x20000024ff5ba062:pn-0x21000024ff5ba062
    
        nvme connect --transport=fc --nqn=nqn.2014-08.org.nvmexpress:nvm-subsystem \
        --traddr=pn-0x200100110d9f4100:nn-0x200100110d9f4100 --host_traddr=nn-0x20000024ff5ba062:pn-0x21000024ff5ba062

Thanks,
Duane

keithbusch, 2016-10-24:
I think this looks fine. Can you send this as a pull request and update the man page in Documentation/nvme-connect.txt?

jpfreyen, 2016-10-24:
Duane,  you will need to update nvme-connect-all.txt as well too as your example shows how to use connect-all (I'm not sure if nvme-connect.txt is even documented).  Which also means 'nvme-discover.txt' needs to be updated because 'connect-all' uses 'discover'.  And you will definitely need to document the '--host_traddr' flag as that is a new flag...the documentation should answer "what is '--host_traddr' providing that --traddr cannot provide for a fc connection, as well as does '--traddr' change it's meaning with respect to a fc connect".  Please also remove "WIP" from 'fc' in the existing documentation (nvme-discover.txt, nvme-connect-all.txt, nvme-connect.txt).

Also, can we use the '-', ie 'host-traddr', over 'host_traddr'?  I think Keith might have mentioned it before that the preference throughout 'nvme-cli' is to not use the underscore, '_'.  Or maybe rename the flag '

jpfreyen, 2016-10-24:
Also, can you please send your nvme-cli changes to the linux-nvme list before getting it merged?  I believe Christoph Hellwig wanted to review the first 'fc' additions to the nvme-cli tool when it was ready.

keithbusch, 2016-10-25:
Thanks for pointing that out, J. Let's keep the option syntax consistent, and get reviews from the community on the linux-nvme mailing list. There are more subscribers to that than the github project.

closed 2019-03-18 by keithbusch