linux-nvme/nvme-cli #138: Add support for Identify data structure with CNS values 02h, 11h and 12h

kpooja88, 2016-11-02:
It would be very helpful if you could add separate commands for Identify data structures for above CNS values, I think what is supported right now is 00h (id-ns), 01h (id-ctrl), 10h (list-ns) and 13h (list-ctrl) as per NVMe spec 1.2. !!

keithbusch, 2016-11-03:
These are all different modes of the existing commands. You get CNS=2 with nvme list-ns --all. You get CNS=11h with nvme id-ns --force. And you get CNS=12h with nvme id-ctrl --namespace-id=<nsid>.

closed 2016-11-03 by keithbusch

kpooja88, 2016-11-03:
Great..thanks!

hellotwozeroonethree, 2016-11-03:
thank you so much来自我的华为平板电脑-------- 原始邮件 --------主题：Re: [linux-nvme/nvme-cli] Add support for Identify data structure with CNS values 02h, 11h and 12h (#138)发件人：Keith Busch 收件人：linux-nvme/nvme-cli 抄送：These are all different modes of the existing commands. You get CNS=2 with nvme list-ns --all. You get CNS=11h with nvme id-ns --force. And you get CNS=12h with nvme id-ctrl --namespace-id=.

  
  

 ript></blockquote></div>