linux-nvme/nvme-cli #139: Get-feature for FID 3 returns raw buffer, but when I specify -b for raw buffer, it returns parsed data.

kpooja88, 2016-11-11:
When i issue get-feature for FID=3, I get a raw buffer out (nvme get-feature -f 3 -n 1)
But when I specify raw binary, I get a parsed output (nvme get-feature -f 3 -n 1 -b)

keithbusch, 2016-11-11:
I've never had an NVMe drive that implemented the LBA Range Type feature. It's the most useless optional feature in the spec, IMHO. You may be the first person I know of to report a test on it, so thanks for raising the issue.

The code, though, looks like the only way to get a parsed output is to use -H. The -b should be unreadable. I'll test this on a virtual controller and get this fixed.

keithbusch, 2016-11-15:
I made a couple minor updates to the output, and it's working as designed now. If you want to get a decoded output, you can use the '-H' option, otherwise it will produce a hex dump. I'm not sure the hex dump by default always makes sense, though. Might need to add a new output format to this command.

closed 2016-11-15 by keithbusch