linux-nvme/nvme-cli #140: NVME List Output JSON Segmentation Fault

Teumer, 2016-11-16:
environment
1. nvme-cli version 1.0
2. ubuntu 14.04.5 kernel 4.1.10

steps to reproduce
1. make && make install
2. nvme list --output-format=json
3. returns "Segmentation fault"
> dmesg
[ 2947.885599] nvme[26026]: segfault at 0 ip 00007f74d1274ebc sp 00007ffd8ffd6820 error 4 in libc-2.19.so[7f74d1206000+1ba000]
[ 2974.819305] nvme[27464]: segfault at 0 ip 00007f31ac4b9ebc sp 00007ffd5c36a410 error 4 in libc-2.19.so[7f31ac44b000+1ba000]
[ 3054.771953] nvme[29480]: segfault at 0 ip 00007f1df528debc sp 00007fff24d03850 error 4 in libc-2.19.so[7f1df521f000+1ba000]

Note other commands formatted w/ json output work w/o issue (only nvme list has the fault)
1. nvme id-ctrl /dev/nvme0 --output-format=json
2. nvme fw-log /dev/nvme0n1 -o json

keithbusch, 2016-11-29:
hm, this is testing successfully here. How many nvme controllers do you have?

Teumer, 2016-12-02:
6 total controllers

keithbusch, 2016-12-06:
Okay, 6 appears to be the tipping point. Will have a fix by today.

keithbusch, 2016-12-06:
oh, 6 was just a red herring on my machine. the real problem is the product name pulled from the models file, which may not be installed on every environment.

keithbusch, 2016-12-06:
Fix pushed.

closed 2016-12-06 by keithbusch