linux-nvme/nvme-cli #143: Cannot compile version 1.0

PatrickVerner, 2016-12-06:
0.9 compiles just fine. I'm running Slackware 14.2.

bash-4.3# make
NVME_VERSION = 1.0
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"1.0"' -c argconfig.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"1.0"' -c suffix.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"1.0"' -c parser.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -std=gnu99 -O2 -g -Wall -Werror -DLIBUDEV_EXISTS -DNVME_VERSION='"1.0"' -c nvme-print.c
nvme-print.c: In function 'json_fw_log':
nvme-print.c:1425:30: error: format '%lu' expects argument of type 'long unsigned int', but argument 4 has type 'long long unsigned int' [-Werror=format=]
   snprintf(str, sizeof(str), "%lu (%s)", (uint64_t)fw_log->frs[i],
                              ^
nvme-print.c:1425:30: error: format '%lu' expects argument of type 'long unsigned int', but argument 4 has type 'long long unsigned int' [-Werror=format=]
cc1: all warnings being treated as errors
Makefile:44: recipe for target 'nvme-print.o' failed
make: *** [nvme-print.o] Error 1
bash-4.3#

keithbusch, 2016-12-06:
I think you must be using a 32-bit compiler. Thanks for the notice, and sorry for the breakage. I'll push a fix today.

keithbusch, 2016-12-06:
Should be fixed in the referenced commit, c77da44.

I also noticed the 48-bit conversion for certain log pages is broken on 32-bit, but happens to truncate such that no warning is emitted. I'll fix that too.

PatrickVerner, 2016-12-07:
Latest git builds now. Thanks!

closed 2016-12-07 by PatrickVerner