linux-nvme/nvme-cli #146: nvme-cli add locate LED、 prepare-for-removal fustion

512868516, 2016-12-15:
hgst nvme manager tool：hdm。
hdm have two fuction that was very useful：
hdm locate               Enables or disables the LED beacon for physical devices.
hdm prepare-for-removal  Prepares physical devices for safe removal from the system

do you have any plan to add locate LED and  prepare-for-removal  fuction？

keithbusch, 2016-12-15:
There's no such thing as an NVMe defined way for LED control. Any such thing would be specific to hardware. If you want to submit a 3rd party plugin for such, that's okay with me.

keithbusch, 2016-12-21:
At the moment, this is simply not an nvme feature. Current LED solutions are vendor and/or platform specific at this time; closing as this tool is not for tracking such issues.

closed 2016-12-21 by keithbusch