linux-nvme/nvme-cli #151: nvme-cli issue

izerkri, 2016-12-21:
Hi,

I'm working on CentOS 7.2 kernel 4.8.8 with the latest nvme-cli version.

After running fio test and executing nvme list sometimes the job gets stuck and we get the following dmesg:
[11457.533097] nvme nvme2: I/O 0 QID 0 timeout, reset controller
[11483.115530] nvme nvme2: Device not ready; aborting reset
[11508.697971] nvme nvme2: Device not ready; aborting reset
[11508.697977] nvme nvme2: Removing after probe failure status: -19
[11508.698359] nvme2n1: detected capacity change from 800166076416 to 0
[11626.482776] nvme nvme3: I/O 0 QID 0 timeout, reset controller
[11652.065242] nvme nvme3: Device not ready; aborting reset
[11677.647674] nvme nvme3: Device not ready; aborting reset
[11677.647717] nvme nvme3: Removing after probe failure status: -19
[11677.648065] nvme3n1: detected capacity change from 800166076416 to 0

after that the drive disappear.
 
Have you seen this issue before?

Thanks,
Ishay.

keithbusch, 2016-12-21:
That indicates your drive did not respond to the identify command the 'nvme list' submits to your drive. The tool can't really help with that, though.

izekri, 2016-12-22:
When working with the same drive on Fedora 20 with kernel 3.19.8, we have never seen this issue.

BruceChar, 2018-02-05:
Hi @izekri 
I got the issue as your case too. My nvme-cli is latest. It seems the issue hasn't been solved yet.
Did you have any way to get through this?
Thank you！

tim-oleksii, 2018-02-05:
I double the answer of Keith - the tool cant help you here.

I've recently submitted a couple of malformed VUCs for some SSD drives causing the behaviour above. The reason was trivial - a misplaced dw11 and dw10. This kind of reset my be incurred by drive firmware assertion or crash, etc.

Whether its OK or not to have an assertion or crash for malformed commands is the other kind of question. At least linux kernel handled this case correctly.

closed 2019-03-18 by keithbusch