linux-nvme/nvme-cli #159: nvme get-ns-id Segmentation fault (core dumped)

yoni386, 2017-02-20:
# nvme get-ns-id
return -> Segmentation fault (core dumped)
dmesg:
nvme[8292]: segfault at 0 ip 00007f6868b4664a sp 00007ffe58f70258 error 4 in libc-2.17.so[7f6868abe000+1b6000]

I think this can be a solution; at /nvme-cli/nvme.c" line 1034 use "get_dev()" instead of "open_dev()"

more info below
static int get_ns_id(int argc, char **argv, struct command *cmd, struct plugin *plugin)
{
        int nsid, fd;
        //fd = open_dev(argv[1]);
        fd = get_dev(argc, argv);
        nsid = nvme_get_nsid(fd);
        if (nsid <= 0) {
                perror(devicename);
                return errno;
        }
        printf("%s: namespace-id:%d\n", devicename, nsid);
        return 0;
}

1031 {
1032         int nsid, fd;
1033         //fd = open_dev(argv[1]);
1034         fd = get_dev(argc, argv);
1035         nsid = nvme_get_nsid(fd);
1036         if (nsid <= 0) {
1037                 perror(devicename);
1038                 return errno;
1039         }
1040         printf("%s: namespace-id:%d\n", devicename, nsid);
1041         return 0;
1042 }

keithbusch, 2017-02-21:
Yah, that would work. Just check that fd < 0 before continuing so that we don't call the ioctl with an invalid file descriptor. You want to send a patch?

closed 2017-02-27 by keithbusch

yoni386, 2017-02-28:
Hello Keith,

The mail got spam. Sorry for not responding.
It might be possible not even checking "fd", On my attemets I tried with
empty valid nvme device and non nvme device and they also work.

Any way, good to know.

Have a nice day,

Yoni Shperling

keithbusch, 2017-02-28:
Without the fd < 0 check, you're relying on the ioctl to handle a bad file descriptor, which it does handle with an appropriate error, but we don't have to make it handle it if we know the file descriptor is invalid.