linux-nvme/nvme-cli #160: Build fail on RHEL6.5

zhenzhuo, 2017-02-23:
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.1"' nvme.c -o nvme argconfig.o suffix.o parser.o nvme-print.o nvme-ioctl.o nvme-lightnvm.o fabrics.o json.o plugin.o intel-nvme.o lnvm-nvme.o memblaze-nvme.o nvme-models.o -luuid
nvme.c:47:23: error: uuid/uuid.h: No such file or directory
nvme.c: In function ‘gen_hostnqn_cmd’:
nvme.c:2680: error: ‘uuid_t’ undeclared (first use in this function)
nvme.c:2680: error: (Each undeclared identifier is reported only once
nvme.c:2680: error: for each function it appears in.)
nvme.c:2680: error: expected ‘;’ before ‘uuid’
cc1: warnings being treated as errors
nvme.c:2683: error: implicit declaration of function ‘uuid_generate_random’
nvme.c:2683: error: ‘uuid’ undeclared (first use in this function)
nvme.c:2684: error: implicit declaration of function ‘uuid_unparse_lower’
make: *** [nvme] Error 1
[root@memory-12 nvme-cli-master]# uname -a
Linux memory-12 2.6.32-431.el6.x86_64 #1 SMP Sun Nov 10 22:19:54 EST 2013 x86_64 x86_64 x86_64 GNU/Linux

keithbusch, 2017-02-23:
I can't see a good way to get around having this dependency unless we want to drop fabrics support. Could you run 'sudo yum install libuuid-devel' and rebuild?

zhenzhuo, 2017-02-24:
Hi Keith,
Thank you for reply.
I built the 22end Dec 2016 release version1.1 on the same server. That version works fine.
Yes, the main different of these 2 versions is uuid.

Best Regards,
zhenzhuo

发送自我的三星 Galaxy 智能手机。

-------- 原始信息 --------
由： Keith Busch <notifications@github.com>
日期: 2017-02-23 23:34 (GMT+08:00)
收件人： linux-nvme/nvme-cli <nvme-cli@noreply.github.com>
抄送： Zhenzhuo Dou <douzhenzhuo@hotmail.com>, Author <author@noreply.github.com>
主题： Re: [linux-nvme/nvme-cli] Build fail on RHEL6.5 (#160)

I can't see a good way to get around having this dependency unless we want to drop fabrics support. Could you run 'sudo yum install libuuid-devel' and rebuild?

keithbusch, 2017-02-24:
So you're okay having this dependency on the uuid library?

zhenzhuo, 2017-02-27:
Sorry for late reply.
How about add macro to nvme.c nvme-builtin.h and modify Makefile.
In some modified linux version,there are no lib uuid-devel.

closed 2017-02-27 by keithbusch

gwendalcr, 2018-10-03:
To allow cros compiling, I would like to go further than fe879af by allowing the caller to enable uuid or not.
It is useful when cross compiling, when the host has libuuid but the target does not want to.

See pull request https://github.com/linux-nvme/nvme-cli/pull/416