linux-nvme/nvme-cli #164: nvme format should default to the lbaf currently in use instead of 0

asciiprod, 2017-03-14:
nvme format defaults to use lba format zero, even if another format is currently in use.
While this can be worked around by first running id-ns before issuing a format, it would be nice, if the lba format would default to the one currently in use.

Seagate XF11440 and XM1440 use format 1 (512e) as default and will change 4Kn when format is run without passing lbaf parameter.

keithbusch, 2017-03-14:
I think that's a good idea if sending a format through the namespace handle or specific NSID. If so, the behavior then will be to use the currently in use LBAF if the user did not specify on in the command line. That sounds reasonable to me.

Not quite as straight forward if sending a format to all namespaces, though.

asciiprod, 2017-03-14:
Yes, basically an "nvme format /dev/nvme0n1" should do the right thing(tm) and format the namespace with the same LBA format that is currently used.

keithbusch, 2017-03-14:
Requested fix pushed to master.

closed 2017-03-14 by keithbusch