linux-nvme/nvme-cli #168: [NVMe-CLI]

asrivastava014, 2017-04-05:
* I am using Null Block device on the Target side for my NVMeF performance benchmarking.
* After Discovery, Connection got established successfully.
* while trying reservation command from Initiator side through nvme-cli...
   #> nvme resv-acquire /dev/nvme0n1 -n 1
I am getting the following error
"NVME IO command error:INVALID_OPCODE(1)"

* nvme list output:
[root@localhost csop]# nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     SPDK00000000000001   SPDK Virtual Controller                  1          67.11  MB /  67.11  MB    512   B +  0 B   FFFFFFFF
/dev/nvme0n2     SPDK00000000000001   SPDK Virtual Controller                  2          67.11  MB /  67.11  MB    512   B +  0 B   FFFFFFFF
/dev/nvme0n3     SPDK00000000000001   SPDK Virtual Controller                  3          67.11  MB /  67.11  MB    512   B +  0 B   FFFFFFFF

Please have a look.

Thanks in advance.

keithbusch, 2017-04-05:
Is the SPDK controller from here? https://github.com/spdk/spdk/blob/master/lib/nvmf/virtual.c#L488

That doesn't support reservations.

keithbusch, 2017-05-08:
Closing; not a tool issue.

If you want SPDK to support this command, will need to submit the request to the vendor that implements the controller interface. AFAIK, the controller you're using doesn't implement multi-controller subsystems, so it seems questionable as to why this command would even be desirable.

closed 2017-05-08 by keithbusch

nama08, 2019-06-04:
Hi,
I am having challenges with NVME format on Samsung drive **(MZVLB256HAHQ).**

**Error:** NVME Admin command error:INVALID_OPCODE(2001)

**Note:** I have been contacting Samsung support, and they weren't able to help because this particular NVMe drive was OEM. When contacted Lenovo /OEM, but no luck there as well as they do not provide support for these drives.