linux-nvme/nvme-cli #171: nvme-cli 1.2 doesn't compile

PatrickVerner, 2017-04-19:
bash-4.3# make
NVME_VERSION = 1.2
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.2"' -c argconfig.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.2"' -c suffix.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.2"' -c parser.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.2"' -c nvme-print.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.2"' -c nvme-ioctl.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.2"' -c nvme-lightnvm.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.2"' -c fabrics.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.2"' -c json.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.2"' -c plugin.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.2"' -c intel-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.2"' -c lnvm-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.2"' -c memblaze-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.2"' -c wdc-nvme.c
wdc-nvme.c: In function 'wdc_dump_length':
wdc-nvme.c:306:19: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
  admin_cmd.addr = (__u64) buf;
                   ^
wdc-nvme.c: In function 'wdc_do_dump':
wdc-nvme.c:342:19: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
  admin_cmd.addr = (__u64) dump_data;
                   ^
wdc-nvme.c: In function 'wdc_do_drive_log':
wdc-nvme.c:482:19: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
  admin_cmd.addr = (__u64) drive_log_data;
                   ^
wdc-nvme.c: In function 'wdc_purge_monitor':
wdc-nvme.c:665:19: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
  admin_cmd.addr = (__u64) output;
                   ^
cc1: all warnings being treated as errors
Makefile:45: recipe for target 'wdc-nvme.o' failed
make: *** [wdc-nvme.o] Error 1
bash-4.3#

keithbusch, 2017-04-19:
You're on a 32-bit machine?

PatrickVerner, 2017-04-19:
Yes.

keithbusch, 2017-04-21:
Fixed in above commit. Thanks for the notification.

closed 2017-04-21 by keithbusch