linux-nvme/nvme-cli #172: get-feature (number of queues) returns six hexadezimal digits

Nirkus, 2017-04-23:
Hey,

while getting to know some Samsung 960 Pro SSDs, I used an Ubuntu 17.04 (zesty, 4.10.0-19 x86_64) live system to issue some `nvme-cli` commands, using package `nvme-cli-1.1-1` from universe.

When geting the number of current/supported queues, command `nvme` prints 6 hexadezimal digits (prefixed by "0x") for the value, while I would have expected 8 for the complete dword (32 bits).

Since the feature is printed as "0x7" instead of 0x07", it seems that this [printf](https://github.com/linux-nvme/nvme-cli/blob/master/nvme.c#L1137-L1139) call in `get_feature` counts the "alternate form" as part of the padding.

On the other hand, `nvme get-feature`prints the following values for number of queues:
- `0x000004`(capatilities)
- `0x070007` (current)

Would have expected the capabilities to be present for both submission and completion queues and for both values to be greater than the current number of queues. ;-)

Gruß, Linus

ikegami-t, 2022-01-04:
The issue should be fixed by the PR changes abobe as mentioned so could you please confirm it with the latest nvme-cli code to close this issue?

igaw, 2022-01-07:
As @ikegami-t indicated this should be fixed by now. Let's close the bug report. If the problem still exists please reopen.

closed 2022-01-07 by igaw