linux-nvme/nvme-cli #174: nvme flush commande

limbo127, 2017-05-03:
hello, 
nvme flush /dev/nvme0n1  or nvme flush /dev/nvme0  is under error with : 
 IO command error:INVALID_NS(b)
I do not understand.

Regards, 
Nicolas

keithbusch, 2017-05-03:
By default, the flush command is sent for "all namespaces". Some controllers don't support that mode (even though they only have one namespace ...), so you have to explicitly set the namespace you want to flush. THat's that "-n" parameter, so something like `nvme flush /dev/nvme0n1 -n 1` should work.

As an enhancement, if you specify a block device, we probably should override the default with the block handle's nsid.

keithbusch, 2017-05-12:
Was the suggestion helpful, or are you still experiencing an issue?

limbo127, 2017-05-12:
Yes it is.
Regard
Nicolas