linux-nvme/nvme-cli #175: smart-log-add command doesn't support json format

locn1000, 2017-05-11:
Hi Keith,
  It seems json output failed with smart-log-add,
    
     [root@udb8562.atn2 ~]# nvme intel smart-log-add /dev/nvme0 -o json
     smart-log-add: unrecognized option '-o'
     Usage: nvme intel smart-log-add <device> [OPTIONS]
    
     Get Intel vendor specific additional smart log (optionally, for the specified
     namespace), and show it.
    
     Options:
       [  --namespace-id=<NUM>, -n <NUM> ]   --- (optional) desired namespace
       [  --raw-binary, -b ]                 --- dump output in binary format
    
     [root@udb8562.atn2 ~]# nvme intel smart-log-add /dev/nvme0
     Additional Smart Log for NVME device:nvme0 namespace-id:ffffffff
     key                               normalized raw
     program_fail_count              : 100%       0
     erase_fail_count                : 100%       0
     wear_leveling                   : 100%       min: 0, max: 0, avg: 0
     end_to_end_error_detection_count: 100%       0
     crc_error_count                 : 100%       0
     timed_workload_media_wear       : 100%       63.999%
     timed_workload_host_reads       : 100%       65535%
     timed_workload_timer            : 100%       65535 min
     thermal_throttle_status         : 100%       0%, cnt: 0
     retry_buffer_overflow_count     : 100%       0
     pll_lock_loss_count             : 100%       0
     nand_bytes_written              : 100%       sectors: 611538
     host_bytes_written              : 100%       sectors: 429062
    
     Thanks,
     Loc Nguyen

keithbusch, 2017-05-12:
We'll get a patch up shortly. Thanks for the notice.

keithbusch, 2017-05-12:
Thanks to Scott for getting this one done.

https://github.com/linux-nvme/nvme-cli/pull/178

closed 2017-05-12 by keithbusch

tim-oleksii, 2018-01-10:
I've just realized that the documentation is not up to date.