linux-nvme/nvme-cli #177: After creating NS I dont see block device on linux "/dev/nvme0n1 "

visheshvijay, 2017-05-12:
[Description]
1) If i create NS and attach to the NVMe controller still I dont see it in block device.
    I have to unload and reload the driver again. Then i see the block device

2) I i want to see the SQ entry sent by the nvme cli command how can i see that ?

[Version]
Kernel version 4.11
NVMe cli 1.2
NVMe driver 1.2

keithbusch, 2017-05-12:
If your controller does not support the NVMe NS Notify Async Event Request, the driver will not automatically rescan for new namespaces. To get the namespace rescan to occur, you can do that with sysfs (echo 1 > /sys/class/nvme/nvme<#>/rescan_controller or echo 1 > /sys/class/nvme/nvme<#>/reset_controller), or through ioctl using NVME_IOCTL_RESCAN or NVME_IOCTL_RESET. If you're controller supports subsystems, you can alternatively use the NVME_IOCTL_SUBSYS_RESET ioctl.

We don't support showing the actual SQ entry. The driver constructs the PRPs and assigns the command ID that are private to the kernel, but you could see the remaining dwords. At the moment, though, the only commands that support that are the "io-passthru" and "admin-passthru" when you append the "--show-command" option.

visheshvijay, 2017-05-12:
Thanks Keith for the update and all useful information.
One more question i have how can i compile nvme 1.0 driver to use always use SGL instead of PRP. I am not sure if this is correct platform to ask this. If i need to post somewhere else let me know.

keithbusch, 2017-05-12:
At least for PCIe NVMe, we don't support SGL at all due to lack of evidence that it's needed. You'd have to write or get a patch if you want to use SGLs.

visheshvijay, 2017-05-13:
Thanks Keith for the valuable inputs. My query has been resolved :)

closed 2017-05-15 by keithbusch

visheshvijay, 2017-05-30:
I have one more query.
Whenever we issue nvme /dev/nvme0 reset we see "resetting controller" print on dmesg but if we send 
nvme /dev/nvme0 subsystem-reset the command passes but there is not print on dmesg. 

How can i see whether nvme inbox driver supports nvme subsystem reset or not ?
I have updated my kernel to 4.11.3 nvme driver version 1.0.

Can you please help to identify whether the NSSR is getting executed or not ? an do we send any message on dmesg

Thanks,
Vishesh

keithbusch, 2017-05-30:
The subsystem-reset command will write "NVMe" to the NSSR register, as per spec, and as a result of that, the drive support subsystems should raise the CSTS.NSSRO flag, which will cause a dmesg to log when the driver detects that. If you're not seeing anything in the logs, your drive doesn't support subsystem resets.