linux-nvme/nvme-cli #184: [NVME-CLI]: Reservation and Dynamic Namespace creation issues with NVMe SSD (PCIe-Based)

asrivastava014, 2017-06-01:
Setup: Host attached NVMe SSD through PCIe

# fdisk -l
Disk /dev/nvme0n1: 1600.3 GB, 1600321314816 bytes, 3125627568 sectors
Units = sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk label type: dos
Disk identifier: 0x2d23c3c9

Device Boot      Start         End      Blocks   Id  System
/dev/nvme0n1p1            2048  1562813783   781405868   83  Linux

I am trying dynamic namespace creation and reservation commands on this physical disk via nvmecli.

Facing the issue:

[root@localhost ~]# nvme create-ns /dev/nvme0n1 -s 52428800 -c 52428800 -f 2 -d 0 -m 1
NVMe Status:INVALID_OPCODE(1)
[root@localhost ~]# nvme create-ns /dev/nvme0n1p1 -s 52428800 -c 52428800 -f 2 -d 0 -m 1
NVMe Status:INVALID_OPCODE(1)

Am I doing anything wrong ?? Please share your thoughts.

keithbusch, 2017-06-01:
Does your device actually support namespace management?

asrivastava014, 2017-06-01:
Is there any way to find out this ??

keithbusch, 2017-06-01:
`nvme id-ctrl /dev/nvme0 | grep oacs`

You're looking to see if bit 3 is set.

asrivastava014, 2017-06-01:
[root@localhost ~]# nvme id-ctrl /dev/nvme0n1 | grep oacs
oacs    : 0x6

Binary: 110 
(3rd Bit is 0)

keithbusch, 2017-06-01:
Closing per submitter confirmation the command is unsupported by the hardware used.

closed 2017-06-01 by keithbusch

asrivastava014, 2017-06-01:
Opps I thinks 0x6 ---->> 1                  1                 0                                                                                                                               
                                         (3rd Bit)      (2nd Bit)      (1st Bit)

keithbusch, 2017-06-01:
Ah, but we start counting at 0. :)

asrivastava014, 2017-06-01:
Optional Admin Command Support (OACS): This field indicates the optional
Admin commands supported by the controller. Refer to section 5.
Bits 15:3 are reserved.

Bit 2 if set to ‘1’ then the controller supports the Firmware Activate and Firmware
Download commands. If cleared to ‘0’ then the controller does not support the
Firmware Activate and Firmware Download commands.

Bit 1 if set to ‘1’ then the controller supports the Format NVM command. If cleared to
‘0’ then the controller does not support the Format NVM command.

Bit 0 if set to ‘1’ then the controller supports the Security Send and Security Receive
commands. If cleared to ‘0’ then the controller does not support the Security Send
and Security Receive commands

So If I am not wrong 0x6 oacs value indicates that it supports NVM Command(Correct if I am wrong)

keithbusch, 2017-06-01:
While that's a very old version of the spec (v1.2 was released 2 years ago!), it's still relevant. Your value of 0x6 says your controller supports NVM Format and Firmware Activate/Download. It does not support anything other optional admin command.

asrivastava014, 2017-06-01:
Thanks a lot for the Help :)