linux-nvme/nvme-cli #185: namespace commands not working on PCIe Based NVME SSD

asrivastava014, 2017-06-02:
[root@localhost ~]# fdisk -l
Disk /dev/nvme0n1: 800.2 GB, 800166076416 bytes, 1562824368 sectors
Units = sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk label type: gpt

[root@localhost ~]# nvme id-ctrl /dev/nvme0n1 | grep oacs
oacs    : 0xe

While Trying to create a 200 GB namespace based on 512b sector 

[root@localhost ~]# nvme create-ns /dev/nvme0n1 -s 419430400 -c 419430400 -f 0 -d 0 -m 1
NVMe Status:NS_INSUFFICIENT_CAPACITY(2115)

Error: Insufficient capacity is flashed, Am I missing something ??
I even tried to reduce nsize and ncap but this error still persists.

Thanks in advance!!

visheshvijay, 2017-06-02:
I guess first you need to delete the NS you already have then try to create and attach it

Thanks,
Vishesh

asrivastava014, 2017-06-02:
Thanks you were right, while playing with the command accidentally a huge namespace was created, which I wasn't aware of.

closed 2017-06-27 by keithbusch