linux-nvme/nvme-cli #187: Need to change the help of firmware activate

visheshvijay, 2017-06-13:
Hi Keith,

According to spec nvme supports commit action 0,1,2,&3.
But in nvme cli help it shows only 0,1,2. But when i tried to give commit action 3 it works :)
We need to update the help commit action : [0-3]

Terminal prints 
[root@jaguar54 26517]# nvme fw-activate /dev/nvme0 -h
Usage: nvme fw-activate <device> [OPTIONS]

Verify downloaded firmware image and commit to specific firmware slot. Device
is not automatically reset following firmware activation. A reset may be
issued with an 'echo 1 > /sys/class/nvme/nvmeX/reset_controller'. Ensure
nvmeX is the device you just activated before reset.

Options:
  [  --slot=<NUM>, -s <NUM> ]           --- firmware slot to activate
  **[  --action=<NUM>, -a <NUM> ]         --- [0-2]: replacement action**

[root@jaguar54 26517]# nvme version
nvme version 1.3
[root@jaguar54 26517]# nvme fw-activate /dev/nvme0 -s 1 -a 3
Success activating firmware action:3 slot:1

Thanks,
Vishesh

keithbusch, 2017-06-13:
Yeah, you're right. Action 3 was reserved back in the day, so neglected to document that one. Will add it, and should probably increase the field to go up to 3 bits for the 1.3 spec updates.

mandarcthorat, 2017-09-13:
Firmware activate help documentation was updated for commit action 3 support by commit dfed0e30aaece49885064a267e6eb2e4f3853ce4. Changes related to NVMe 1.3 spec are still pending

closed 2019-03-18 by keithbusch