linux-nvme/nvme-cli #191: nvme connect -k  -c is not working

512868516, 2017-06-18:
```
[root@sto3 ~]# nvme connect -h
Usage: nvme connect <device> [OPTIONS]

Connect to NVMeoF subsystem

Options:
  [  --keep-alive-tmo=<LIST>, -k <LIST> ] --- keep alive timeout period in
                                            seconds
  [  --reconnect-delay=<LIST>, -c <LIST> ] --- reconnect timeout period in
                                            seconds
```

on the client i execute
```
nvme connect -t rdma -a 172.16.130.1 -s 4421 -n nvme1n1 -k 6 -c 6
```
then i run fio random write test,during the test  i pull out my ib connection between the client and the server.
but the fio  will hang a lone time ,not return io error, the -k and -c par seems not working.

512868516, 2017-06-22:
@keithbusch

keithbusch, 2017-06-22:
Sounds like a kernel or driver problem. Want to take this one to the linux-nvme mailing list?

512868516, 2017-06-23:
ok

closed 2017-08-04 by 512868516