linux-nvme/nvme-cli #201: nvme lnvm not working

nusbaoning, 2017-06-29:
According to http://lightnvm.io/pblk-tools/tools.html, the NVMe CLI tool supports Open-Channel SSDs through the lnvm extension.
However, I installed nvme-cli and also lnvm, but nvme lnvm does not work. It shows unknown command "lnvm".
How can I let nvme-cli work with lnvm extension of Open-Channel SSDs so that I can follow the intro of http://lightnvm.io/pblk-tools/tools.html to expose a block device?
Thank you very much!

keithbusch, 2017-06-29:
What version are you running? If I run "nvme lnvm" I get the following output:

```
# nvme lnvm
nvme-1.3.31.g7544
usage: nvme lnvm <command> [<device>] [<args>]

The '<device>' may be either an NVMe character device (ex: /dev/nvme0) or an
nvme block device (ex: /dev/nvme0n1).

LightNVM specific extensions

The following are all implemented sub-commands:
  list            List available LightNVM devices
  info            List general information and available target engines
  id-ns           List geometry for LightNVM device
  init            Initialize media manager on LightNVM device
  create          Create target on top of a LightNVM device
  remove          Remove target from device
  factory         Reset device to factory state
  diag-bbtbl      Diagnose bad block table
  diag-set-bbtbl  Update bad block table
  version         Shows the program version
  help            Display this help

See 'nvme lnvm help <command>' for more information on a specific command
```

nusbaoning, 2017-06-30:
My nvme version is 0.5. 
I installed it by sudo apt-get install. I will download it from git and try.
Thank you.

nusbaoning, 2017-06-30:
Sorry when I try to use the Ubuntu PPA to install nvme-cli, I can only install the 0.5 version. 
[image]
If I download the git and use make & make install, there is no error message but nvme is not installed.
[image]
[image]

MatiasBjorling, 2017-06-30:
When doing make; make install, it is installed into /usr/local

If you want a global install, you can use PREFIX=/usr make; make install (you might also have to prefix make install)

That'll install it globally, but it will mess if you install the nvme package later.

nusbaoning, 2017-06-30:
I see. It now works! Thank you so much!

closed 2017-06-30 by nusbaoning