linux-nvme/nvme-cli #202: get-feature does not support mandatory features

narasimhan-v, 2017-06-29:
I have an NVMe controller which does not support any of the mandatory feature, as mentioned by https://github.com/linux-nvme/nvme-cli/blob/master/tests/nvme_get_features_test.py

Only `0x3` is supported, which is `LBA Range Type`

Please clarify where to get the list of supported features for a controller.

keithbusch, 2017-06-29:
Section 5.21.1 of the NVMe specification 1.3 (http://www.nvmexpress.org/wp-content/uploads/NVM_Express_Revision_1.3.pdf) has a list of all the mandatory and optional features. Interestingly enough, your controller implements an optional feature but sounds like it neglects to implement mandatory ones.

narasimhan-v, 2017-06-29:
Thanks Keith, will check with the controller vendor.

closed 2017-06-29 by keithbusch